microsoft/ELL #106: Image Classification Step 3 fails

IvanFarkas, 2017-11-01:
I built ELL successfully with CUDA and OpenBLAS (since I fixed CMake/OpenBLASSetup.cmake to support model 158). Ran test successfully (attached).

Image Classification Step 1, 2 OK.
In Step 3 the following command fails.
```shell
cmake -G "Visual Studio 15 2017 Win64" .. && cmake --build . --config release && cd ..\..
```
Error:
```
model.obj : error LNK2019: unresolved external symbol cblas_sgemm referenced in function model_Predict [C:\ML\EEL\Tutorial\Image_Classification\host\build\_model.vcxproj]
C:\ML\EEL\Tutorial\Image_Classification\host\build\Release\_model.pyd : fatal error LNK1120: 1 unresolved externals [C:\ML\EEL\Tutorial\Image_Classification\host\build\_model.vcxproj]
```
[Error.txt](https://github.com/Microsoft/ELL/files/1435604/Error.txt)
[Test Results.txt](https://github.com/Microsoft/ELL/files/1435562/Test.Results.txt)

lovettchris, 2017-11-10:
Assuming you followed the [INSTALL-windows  ](https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md) instructions you should have the openblas library in the external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\lib.  If that is not there you may have missed this step:
```
nuget.exe restore external/packages.config -PackagesDirectory external
```
and you then may get additional help from the **Path Environment** steps at the end of the setup page.

IvanFarkas, 2017-11-10:
Without CUDA and OpenBLAS is OK.
I'll repeat the test when proper processor model detection (OpenBLAS) and **VS 2017** build with **CUDA 9** fix is in.

IvanFarkas, 2017-11-29:
Re-tested. Success.

closed 2017-11-29 by IvanFarkas