microsoft/ELL #107: WSL Ubuntu - Test 14: cntk_importer_test fails

IvanFarkas, 2017-11-01:
I followed the **Ubuntu** (WSL) installation instructions.
All test, but one ran successfully. **Test 14: cntk_importer_test** failed.
```
test 14
      Start 14: cntk_importer_test

14: Test command: /home/ifarkas/miniconda3/envs/py36/bin/python "-m" "unittest" "cntk_to_ell_importer_test.py" "cntk_full_model_test.py"
14: Test timeout computed to be: 9.99988e+06
14: ssssssssssE
14: ======================================================================
14: ERROR: cntk_full_model_test (unittest.loader._FailedTest)
14: ----------------------------------------------------------------------
14: ImportError: Failed to import test module: cntk_full_model_test
14: Traceback (most recent call last):
14:   File "/home/ifarkas/miniconda3/envs/py36/lib/python3.6/unittest/loader.py", line 153, in loadTestsFromName
14:     module = __import__(module_name)
14:   File "/home/ifarkas/ELL/build/tools/importers/CNTK/test/cntk_full_model_test.py", line 10, in <module>
14:     import cntk
14: ModuleNotFoundError: No module named 'cntk'
14:
14:
14: ----------------------------------------------------------------------
14: Ran 11 tests in 0.001s
14:
14: FAILED (errors=1, skipped=10)
```

kernhanda, 2017-11-02:
Per our [importing models](https://microsoft.github.io/ELL/tutorials/Importing-models/) tutorial, please install CNTK on your machine.

IvanFarkas, 2017-11-02:
I am waiting for **CNTK** **CUDA 9** and **VS 2017** compatibility.
It's promised that it will be done in this iteration, hopefully in a couple of days. Thx.

kernhanda, 2017-11-02:
If you're on WSL, you don't need the VS 2017 compatibility.

IvanFarkas, 2017-11-02:
Correct. Confused with the other similar issue. I'll just wait for CUDA 9 support. Thx.

IvanFarkas, 2017-11-30:
It works now. All tests succeded.
Installed CNTK 2.3
ran this

```shell
source activate py36
cd ELL/build
ctest --force-new-ctest-process --verbose -C Release
```

**Result is OK**
```shell
test 14
      Start 14: cntk_importer_test

14: Test command: /home/ifarkas/miniconda3/envs/py36/bin/python "-m" "unittest" "cntk_to_ell_importer_test.py"
14: Test timeout computed to be: 9.99988e+06
14: ssssssssss
14: ----------------------------------------------------------------------
14: Ran 10 tests in 0.000s
14:
14: OK (skipped=10)
14/33 Test #14: cntk_importer_test ...............   Passed    0.33 sec
```

closed 2017-11-30 by IvanFarkas