microsoft/ELL #108: Test 9 - model_compiler_test, test 14 - cntk_importer_test failed

IvanFarkas, 2017-11-02:
I compiled bare bone ELL (no CUDA nor OpenBLAS) on Windows 10 Pro 10.0.16299 x64, VS 2017 15.4.2.
**Test 9 - model_compiler_test** and **test 14 - cntk_importer_test** failed.

**Test 9**
```
test 9
      Start  9: model_compiler_test
9: Test command: C:\ML\EEL\ELL\build\libraries\model\Release\model_compiler_test.exe
9: Environment variables:
9:  PATH=C:/ML/EEL/ELL/build/bin/Release/
9:  %PATH%
9: Test timeout computed to be: 9.99988e+006
9: ERROR, got ELL exception. Message: llvm fatal error: Program used external function 'cblas_sgemv' which could not be resolved!
 9/35 Test  #9: model_compiler_test ..............***Failed    0.17 sec
```

**Test 14**
```
test 14
      Start 14: cntk_importer_test
14: Test command: C:\Users\IvanF\AppData\Local\conda\conda\envs\py36\python.exe "-m" "unittest" "cntk_to_ell_importer_test.py" "cntk_full_model_test.py"
14: Test timeout computed to be: 9.99988e+006
14: ssssssssssE
14: ======================================================================
14: ERROR: cntk_full_model_test (unittest.loader._FailedTest)
14: ----------------------------------------------------------------------
14: ImportError: Failed to import test module: cntk_full_model_test
14: Traceback (most recent call last):
14:   File "C:\Users\IvanF\AppData\Local\conda\conda\envs\py36\lib\unittest\loader.py", line 153, in loadTestsFromName
14:     module = __import__(module_name)
14:   File "C:\ML\EEL\ELL\build\tools\importers\CNTK\test\cntk_full_model_test.py", line 10, in <module>
14:     import cntk
14: ModuleNotFoundError: No module named 'cntk'
14:
14:
14: ----------------------------------------------------------------------
14: Ran 11 tests in 0.000s
14:
14: FAILED (errors=1, skipped=10)
```

kernhanda, 2017-11-02:
Per our [importing models](https://microsoft.github.io/ELL/tutorials/Importing-models/) tutorial, please install CNTK on your machine.

IvanFarkas, 2017-11-02:
I am waiting for **CNTK** **CUDA 9** and **VS 2017** compatibility.
It's promised that it will be done in this iteration, hopefully in a couple of days. Thx.

IvanFarkas, 2017-11-29:
Installed CNTK 2.3. All tests passed.
[Test Results.txt](https://github.com/Microsoft/ELL/files/1514318/Test.Results.txt)

closed 2017-11-29 by IvanFarkas