microsoft/ELL #110: Importing models tutorial - CNTK model conversion fails

IvanFarkas, 2017-11-04:
Installed CNTK 2.2 CPU only python package.
DarkNet to ELL conversion works OK.
**CNTK** to **ELL** conversion fails.
Which version of CNTK was used for the model in this tutorial?
Could this be a version issue?

Ran the command below.
```
python <ELL-root>/tools/importers/cntk/cntk_import.py model.cntk
```
Getting a **CNTK** to **ELL** conversion error (complete output attached).
```
Error occurred attempting to convert cntk layers to ELL layers
Traceback (most recent call last):
  File "C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py", line 59, in <module>
    main(sys.argv[1:]) # drop the first argument (program name)
  File "C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py", line 44, in main
    predictor = cntk_to_ell.predictor_from_cntk_model(filename)
  File "C:\ML\EEL\ELL\tools\importers\cntk\cntk_to_ell.py", line 53, in predictor_from_cntk_model
    raise exception
  File "C:\ML\EEL\ELL\tools\importers\cntk\cntk_to_ell.py", line 50, in predictor_from_cntk_model
    predictor = ELL.FloatNeuralNetworkPredictor(ellLayers)
  File "C:\ML\EEL\ELL\build\interfaces\python\ELL.py", line 4448, in __init__
    this = _ELL.new_FloatNeuralNetworkPredictor(layers, scaleFactor)
ValueError: Input tensor must not exceed output tensor (minus padding) dimensions for activation layer.
```

[CNTK to ELL Error.txt](https://github.com/Microsoft/ELL/files/1443100/CNTK.to.ELL.Error.txt)

kernhanda, 2017-11-09:
Unfortunately, ELL does not support converting CNTK's `Combine` layers.

lisaong, 2017-11-09:
Hi Ivan,

What model did you use for the import? 

The error is coming from line 36:
https://github.com/Microsoft/ELL/blob/a314974e8ba180132253ddbc7b4a9aed8e17fde4/libraries/predictors/neural/tcc/ActivationLayer.tcc

From the error text file, this is a different model architecture from the VGG16_ImageNet_Caffe.model referenced in the tutorial.  Maybe one of the RELU layers in the model you've used is not being imported correctly.

IvanFarkas, 2017-11-09:
C:/ML/EEL/ELL is the ELL GitHup repo root.

lisaong, 2017-11-09:
I should have been clearer, the cntk.model that you used has this architecture:

```
Convolution : 226x226x3 -> 224x224x64 | input padding 1 output padding 0
ReLU : 224x224x64 -> 226x226x64 | input padding 0 output padding 1
Convolution : 226x226x64 -> 224x224x64 | input padding 1 output padding 0
ReLU : 224x224x64 -> 224x224x64 | input padding 0 output padding 0
MaxPooling : 224x224x64 -> 114x114x64 | input padding 0 output padding 1
Convolution : 114x114x64 -> 112x112x128 | input padding 1 output padding 0
ReLU : 112x112x128 -> 114x114x128 | input padding 0 output padding 1
Convolution : 114x114x128 -> 112x112x128 | input padding 1 output padding 0
ReLU : 112x112x128 -> 112x112x128 | input padding 0 output padding 0
MaxPooling : 112x112x128 -> 58x58x128 | input padding 0 output padding 1
Convolution : 58x58x128 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 58x58x256 | input padding 0 output padding 1
Convolution : 58x58x256 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 58x58x256 | input padding 0 output padding 1
Convolution : 58x58x256 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 56x56x256 | input padding 0 output padding 0
MaxPooling : 56x56x256 -> 30x30x256 | input padding 0 output padding 1
Convolution : 30x30x256 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 30x30x512 | input padding 0 output padding 1
Convolution : 30x30x512 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 30x30x512 | input padding 0 output padding 1
Convolution : 30x30x512 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 28x28x512 | input padding 0 output padding 0
MaxPooling : 28x28x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 14x14x512 | input padding 0 output padding 0
MaxPooling : 14x14x512 -> 7x7x512 | input padding 0 output padding 0
ReLU : 1x1x4096 -> 1x1x4096 | input padding 0 output padding 0
ReLU : 1x1x4096 -> 1x1x4096 | input padding 0 output padding 0
Softmax : 1x1x1000 -> 1x1x1000 | input padding 0 output padding 0
```

Is your cntk.model the downloaded VGG16_ImageNet_Caffe.model, or another model?

lisa

IvanFarkas, 2017-11-09:
The full output is below

```
(py36) C:\ML\EEL\Tutorial\Importing_Models>python C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py model.cntk
Loading...
Selected CPU as the process wide default device.

Finished loading.
Pre-processing...

Will not process linear - skipping this layer as irrelevant.

Will not process Dropout - skipping this layer as irrelevant.

Will not process linear - skipping this layer as irrelevant.

Will not process Dropout - skipping this layer as irrelevant.

Will not process linear - skipping this layer as irrelevant.

Will not process Combine - skipping this layer as irrelevant.
Convolution : 226x226x3 -> 224x224x64 | input padding 1 output padding 0
ReLU : 224x224x64 -> 226x226x64 | input padding 0 output padding 1
Convolution : 226x226x64 -> 224x224x64 | input padding 1 output padding 0
ReLU : 224x224x64 -> 224x224x64 | input padding 0 output padding 0
MaxPooling : 224x224x64 -> 114x114x64 | input padding 0 output padding 1
Convolution : 114x114x64 -> 112x112x128 | input padding 1 output padding 0
ReLU : 112x112x128 -> 114x114x128 | input padding 0 output padding 1
Convolution : 114x114x128 -> 112x112x128 | input padding 1 output padding 0
ReLU : 112x112x128 -> 112x112x128 | input padding 0 output padding 0
MaxPooling : 112x112x128 -> 58x58x128 | input padding 0 output padding 1
Convolution : 58x58x128 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 58x58x256 | input padding 0 output padding 1
Convolution : 58x58x256 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 58x58x256 | input padding 0 output padding 1
Convolution : 58x58x256 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 56x56x256 | input padding 0 output padding 0
MaxPooling : 56x56x256 -> 30x30x256 | input padding 0 output padding 1
Convolution : 30x30x256 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 30x30x512 | input padding 0 output padding 1
Convolution : 30x30x512 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 30x30x512 | input padding 0 output padding 1
Convolution : 30x30x512 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 28x28x512 | input padding 0 output padding 0
MaxPooling : 28x28x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 14x14x512 | input padding 0 output padding 0
MaxPooling : 14x14x512 -> 7x7x512 | input padding 0 output padding 0
ReLU : 1x1x4096 -> 1x1x4096 | input padding 0 output padding 0
ReLU : 1x1x4096 -> 1x1x4096 | input padding 0 output padding 0
Softmax : 1x1x1000 -> 1x1x1000 | input padding 0 output padding 0

Finished pre-processing.
Error occurred attempting to convert cntk layers to ELL layers
Traceback (most recent call last):
  File "C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py", line 59, in <module>
    main(sys.argv[1:]) # drop the first argument (program name)
  File "C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py", line 44, in main
    predictor = cntk_to_ell.predictor_from_cntk_model(filename)
  File "C:\ML\EEL\ELL\tools\importers\cntk\cntk_to_ell.py", line 53, in predictor_from_cntk_model
    raise exception
  File "C:\ML\EEL\ELL\tools\importers\cntk\cntk_to_ell.py", line 50, in predictor_from_cntk_model
    predictor = ELL.FloatNeuralNetworkPredictor(ellLayers)
  File "C:\ML\EEL\ELL\build\interfaces\python\ELL.py", line 4448, in __init__
    this = _ELL.new_FloatNeuralNetworkPredictor(layers, scaleFactor)
ValueError: Input tensor must not exceed output tensor (minus padding) dimensions for activation layer.
```

IvanFarkas, 2017-11-09:
I do not have CNTK installed.
I am still waiting to be able to compile with **VS 2017**.

lisaong, 2017-11-10:
Where did you download your model.cntk file?

IvanFarkas, 2017-11-10:
I am just following your [playbook](https://microsoft.github.io/ELL/tutorials/Importing-models/).

**Step1 - Get CNTK model VGG16_ImageNet_Caffe.model**
```
(py36) C:\ML\EEL\Tutorial\Importing_Models>curl --location -o model.cntk https://www.cntk.ai/Models/Caffe_Converted/VGG16_ImageNet_Caffe.model
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  527M  100  527M    0     0  5657k      0  0:01:35  0:01:35 --:--:-- 4610k
```

Step 2 - Get categories.txt
```
(py36) C:\ML\EEL\Tutorial\Importing_Models>curl --location -o categories.txt https://raw.githubusercontent.com/Microsoft/ELL-models/master/models/ILSVRC2012/ILSVRC2012_labels.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    15  100    15    0     0    120      0 --:--:-- --:--:-- --:--:--   120
```

**Step 3 - Run cntk_import.py**
```
(py36) C:\ML\EEL\Tutorial\Importing_Models>python C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py model.cntk
Loading...
Selected CPU as the process wide default device.

Finished loading.
Pre-processing...

Will not process linear - skipping this layer as irrelevant.

Will not process Dropout - skipping this layer as irrelevant.

Will not process linear - skipping this layer as irrelevant.

Will not process Dropout - skipping this layer as irrelevant.

Will not process linear - skipping this layer as irrelevant.

Will not process Combine - skipping this layer as irrelevant.
Convolution : 226x226x3 -> 224x224x64 | input padding 1 output padding 0
ReLU : 224x224x64 -> 226x226x64 | input padding 0 output padding 1
Convolution : 226x226x64 -> 224x224x64 | input padding 1 output padding 0
ReLU : 224x224x64 -> 224x224x64 | input padding 0 output padding 0
MaxPooling : 224x224x64 -> 114x114x64 | input padding 0 output padding 1
Convolution : 114x114x64 -> 112x112x128 | input padding 1 output padding 0
ReLU : 112x112x128 -> 114x114x128 | input padding 0 output padding 1
Convolution : 114x114x128 -> 112x112x128 | input padding 1 output padding 0
ReLU : 112x112x128 -> 112x112x128 | input padding 0 output padding 0
MaxPooling : 112x112x128 -> 58x58x128 | input padding 0 output padding 1
Convolution : 58x58x128 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 58x58x256 | input padding 0 output padding 1
Convolution : 58x58x256 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 58x58x256 | input padding 0 output padding 1
Convolution : 58x58x256 -> 56x56x256 | input padding 1 output padding 0
ReLU : 56x56x256 -> 56x56x256 | input padding 0 output padding 0
MaxPooling : 56x56x256 -> 30x30x256 | input padding 0 output padding 1
Convolution : 30x30x256 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 30x30x512 | input padding 0 output padding 1
Convolution : 30x30x512 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 30x30x512 | input padding 0 output padding 1
Convolution : 30x30x512 -> 28x28x512 | input padding 1 output padding 0
ReLU : 28x28x512 -> 28x28x512 | input padding 0 output padding 0
MaxPooling : 28x28x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 16x16x512 | input padding 0 output padding 1
Convolution : 16x16x512 -> 14x14x512 | input padding 1 output padding 0
ReLU : 14x14x512 -> 14x14x512 | input padding 0 output padding 0
MaxPooling : 14x14x512 -> 7x7x512 | input padding 0 output padding 0
ReLU : 1x1x4096 -> 1x1x4096 | input padding 0 output padding 0
ReLU : 1x1x4096 -> 1x1x4096 | input padding 0 output padding 0
Softmax : 1x1x1000 -> 1x1x1000 | input padding 0 output padding 0

Finished pre-processing.
Error occurred attempting to convert cntk layers to ELL layers
Traceback (most recent call last):
  File "C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py", line 59, in <module>
    main(sys.argv[1:]) # drop the first argument (program name)
  File "C:/ML/EEL/ELL/tools/importers/cntk/cntk_import.py", line 44, in main
    predictor = cntk_to_ell.predictor_from_cntk_model(filename)
  File "C:\ML\EEL\ELL\tools\importers\cntk\cntk_to_ell.py", line 53, in predictor_from_cntk_model
    raise exception
  File "C:\ML\EEL\ELL\tools\importers\cntk\cntk_to_ell.py", line 50, in predictor_from_cntk_model
    predictor = ELL.FloatNeuralNetworkPredictor(ellLayers)
  File "C:\ML\EEL\ELL\build\interfaces\python\ELL.py", line 4448, in __init__
    this = _ELL.new_FloatNeuralNetworkPredictor(layers, scaleFactor)
ValueError: Input tensor must not exceed output tensor (minus padding) dimensions for activation layer.
```

lisaong, 2017-11-10:
Okay, will take a look.

lisaong, 2017-11-10:
Thanks for reporting this! Fix is on the way, but if you don't feel like waiting, here's where to make the change:

ELL/tools/importers/CNTK/lib/cntk_layers.py, around line 922:

< elif (cntkLayer.op_name == 'Linear'):
> elif (cntkLayer.op_name == 'linear'):  # this op_name must be lower-case

lisaong, 2017-11-13:
Fixed in v2.0.5

closed 2017-11-13 by lisaong

IvanFarkas, 2017-11-13:
I'll retest by tomorrow. Thx Lisa.