microsoft/ELL #111: What's the ETA for Raspbian Stretch compatibility?

IvanFarkas, 2017-11-07:
What's the ETA for **Raspbian Stretch** compatibility?

lovettchris, 2017-11-10:
ELL itself is fine on Stretch, the only problem is finding a good opencv solution.  You may have noticed we built a custom opencv for Raspberry Pi and published it in our microsoft-ell conda channel.

Stretch updated a lot of things, (GCC 6, and new avcodecs) unfortunately this breaks our existing microsoft-ell opencv package because that build of opencv assumes the Jessie version of libavcodec.so.56, but Stretch is now using libavcodec.so.57.  This would normally not be a problem except this is a built in package included in the Raspbian image and those cannot easily be down leveled, at least I couldn’t find an easy way to do it…nor could I find another apt-get installable package that includes the older version.  

So the work around for now if you need opencv on Stretch is to build it yourself.  You can build it on the pi, It takes a long time, but it can be done.  You'll want to build it with this command line to ensure you don't run out of space or memory:
```
	cmake -DBUILD_SHARED_LIBS=OFF -DBUILD_TESTS=OFF -DBUILD_PERF_TESTS=OFF \
	-DPYTHON3_EXECUTABLE="/usr/bin/python3.4" \
	-DPYTHON3_LIBRARY="/usr/lib/python3.4" \
	-DPYTHON3_INCLUDE_DIR="/usr/include/python3.4" \
        -DCMAKE_BUILD_TYPE="Release" -Wno-dev ..
```

IvanFarkas, 2017-11-10:
That is so wonderful.
I did that already as a separate exercise based on this great article:
**[Optimizing OpenCV on the Raspberry Pi](https://www.pyimagesearch.com/2017/10/09/optimizing-opencv-on-the-raspberry-pi/)** by [Adrian Rosebrock](https://www.pyimagesearch.com/author/adrian/) 

It would be great to include **Optimizing OpenCV on the Raspberry Pi** in your resources as an advanced material.
Also, it would be fantastic if you share your roadblocks or blocking dependencies with the community so motivated developers like myself can help all of us to succeed faster.

I followed the article with slight modifications:

1. I skipped Step #4, since I had MiniConda installed that ELL is using,
2. Added the following to CMake:
```
  -D PYTHON3_PACKAGES_PATH=/home/ifarkas/miniconda3/envs/py34/lib/python3.4/site-packages \
  -D PYTHON3_LIBRARY=/home/ifarkas/miniconda3/envs/py34/lib/libpython3.4m.so \
  -D PYTHON_INCLUDE_DIR=/home/ifarkas/miniconda3/envs/py34/include/python3.4m \
  -D PYTHON3_EXECUTABLE=/home/ifarkas/miniconda3/envs/py34/bin/python3.4 \
  -D PYTHON_NUMPY_INCLUDE_DIRS=/home/ifarkas/miniconda3/envs/py34/lib/python3.4/site-packages/numpy/core/include 
```

lovettchris, 2017-11-10:
You are right, that is an excellent article by Adrian Rosebrock, thanks for the link.  And congratulations for being a motivated developer, so you got opencv working on Stretch with py34 bindings ?

IvanFarkas, 2017-11-10:
Not yet. I did the optimization on Jessie with success.
I'll do Stretch with py34 bindings over the weekend.

IvanFarkas, 2017-11-10:
I got super fast SDs and will try 2 different **"Best"** cooling and overclock strategies.
I can't wait to share the fast and OpenCV optimized test results.
After success, I'll do the same on DragonBoard 410c.

lovettchris, 2017-11-10:
That's great, hey if you are interested, it would be awesome if you published your results on your own conda channel so that we can point ELL customers to your version of OpenCV for Stretch.

IvanFarkas, 2017-11-11:
Working on it

lovettchris, 2017-11-11:
awesome!

IvanFarkas, 2017-11-15:
**[Embedded Learning Library (ELL) – Getting Started](http://ivan-farkas.com/2017/11/14/ell-getting-started/)** is published.

lovettchris, 2017-11-15:
Wow you have been busy! Fantastic blog.

IvanFarkas, 2017-11-15:
Thx Chris.
Conda channel is coming with a highly **optimized OpenCV** for **Raspberry Pi Raspbian Stretch**.
I'll update the blog when it's ready.

IvanFarkas, 2017-11-16:
I am new to creating a Conda armv7l (Raspbian) package recipe for OpenCV.
Based on the **[Building conda packages for general code projects](https://conda.io/docs/user-guide/tutorials/build-postgis.html)** Conda tutorial does not seem too complicated, but as we all know the devil is always in the details.

I used **[opencv-python](https://anaconda.org/conda-forge/opencv)** packages on conda-forge channel as a base.
Unzipped [linux-64](https://anaconda.org/conda-forge/opencv/files/modal/info/5a099a94a9fe2a1203a0bd70) [opencv-3.3.0-py36_blas_openblas_201.tar.bz2](https://anaconda.org/conda-forge/opencv/3.3.0/download/linux-64/opencv-3.3.0-py36_blas_openblas_201.tar.bz2) and used \info\recipe\meta.yaml.template as my meta.yaml recipe (attached).
Due to the fact that armv7l does not get the same attention as Windows, Linux and MacOS make it nearly impossible to accomplish this task.
I found some armv7l OpenCV packages that are half-baked like **[opencv-python-armv7l 3.2.0](https://pypi.python.org/pypi/opencv-python-armv7l/3.2.0)** (no FFmpeg and cv2.imshow() does not work) (see METADATA.txt), and that is just unexaptable.

It's entirely possible that I am missing something. Community! SOS!

(lose the .txt extension)
[meta.yaml.txt](https://github.com/Microsoft/ELL/files/1478587/meta.yaml.txt)
[METADATA.txt](https://github.com/Microsoft/ELL/files/1478618/METADATA.txt)

lisaong, 2017-11-16:
@IvanFarkas, we use this repo to build opencv conda packages for armv71 (Jessie).  You can fork it as a starting point.  The opencv3 folder contains a build script that you can either customize or fork.  

https://github.com/lisaong/raspberrypi-conda-recipes

IvanFarkas, 2017-11-29:
@lisaong I tried and I ran into dependency hell. I'll revisit another time. Thx for your help Lisa.

closed 2017-11-29 by IvanFarkas