microsoft/ELL #112: Running speed of darknet model

Tryljs, 2017-11-10:
Dear team
Does the darknet model run with the ELL predictor faster than run with the darknet predictor?

lovettchris, 2017-11-10:
If using only the CPU to do the job yes.  Darknet's inference takes 1.49 seconds using [this imagenet model](https://pjreddie.com/darknet/imagenet), whereas the same model compiled by ELL takes 94 milliseconds.

But if you compile darknet with GPU then it will be faster than ELL.  ELL does not support targeting GPU as it is designed for small embedded hardware that does not have a big GPU.

For example, ELL can run the same darknet model on a raspberry pi 3 in 3.6 seconds.  If you don't care so much about the accuracy of the model you can get simpler models to run much faster on the pi, as shown in the [model gallery](https://microsoft.github.io/ELL/gallery/).

Tryljs, 2017-11-10:
Thank you very much.

closed 2018-07-17 by lovettchris