microsoft/ELL #114: ELL vs AI Toolkit for Azure IoT Edge

IvanFarkas, 2017-11-15:
As I am watching Connect() I am somewhat confused about the segmented MS AI landscape.
Can MS have a comparison chart with a bunch of use cases which ML tool use where.
Specifically, I am very confused about how **ELL** and **AI Toolkit for Azure IoT Edge** differs.
Thanks, guys. Keep up the great work.

oferdekel, 2017-11-16:
Microsoft strives to make great products for our customers – Azure and Azure IoT are good examples. One way to create best-in-class products is to invest in basic computer science research, which is what we do at Microsoft Research. ELL is not a Microsoft product, it is an early preview of the ongoing research work done at Microsoft Research around resource-constrained AI. Microsoft Research often shares its latest research innovations with the external research community (and with tech enthusiasts in general) by publishing research papers and releasing open source code.

IvanFarkas, 2017-11-24:
That's awesome, but it still does not answer my question.
How does **ELL** and **AI Toolkit for Azure IoT Edge** differ, please?

eric-tang, 2017-11-27:
@IvanFarkas I'm prototyping a little project which gets both involved. Here is my understanding of the difference between AI Toolkit for Azure IoT edge and ELL:

The most notable thing to understand is the difference between the "Edge" concept in the term "Azure IoT Edge" and the "thing" concept in IoT area. If you go through the tutorial of Azure IoT edge, the "Edge" is actually a reasonable powerful local computing resource, which could be a Linux VM, a powerful local workstation or even a group of private local cloud deployed by your own. The Azure ToolKit (IMHO, it's actually not limited to just IoT kit, it could be any Azure toolset) is to deploy a bunch of Azure tools (e.g. some Machine Learning Framework/Libraries) onto your Edge devices which makes your Edge device a gateway between your IoT devices and Azure Cloud. The purposes for the concept of "Edge" is multi-fold, some straightforward benefits are: (1) Minimize the latency of the decision making between your IoT devices. You "Edge" device can quickly make the decision based on the information collected from IoT things and act on that fast by deliver the instruction on a local network. This is critical in some scenarios such as Drone/Robot control; (2) Minimize your cost for directly connect your IoT devices to Azure cloud. Imagine you have millions of IoT nodes in the wild and all of them keep sending messages to Azure Cloud. It would intrigue significant amount of cost on your bill. With a edge device, it can collect all those messages, analyze it, compress it and send it in a more efficient way to Azure cloud to minimize your cost. You can compare Azure IoT Edge to AWS GreenGrass service to understand the reasoning behind.

On the other hand, ELL is an AI focused library intended to run learning algorithms on resources constrained systems, such as a Raspberry Pi in the tutorial (Actually I think the confusion might come from the Raspberry Pi, since this little device is powerful enough to run as a "Edge"). The purpose of this library is to help you use the advanced AI techniques on you local IoT without Network with a reasonable amount of power consumption. My current prototype project requires to recognize some wild animals in the wild with no internet or very limited cellular connection. I cannot take pictures and consistently send large pictures back for recognition. Then ELL is super critical to me since I need to run it on my local IoT device and just send the recognition result back to my main service (either Edge or Azure public cloud).

Hopefully my understanding is a correct interpretation here and inspires the discussion here. @oferdekel

IvanFarkas, 2017-11-27:
@oferdekel Thank you very much for the quick and detailed explanation.
I do understand the differences now.
For Edge computing, I am looking for more powerful SBCs like DragonBoard 820c and the mighty Snapdragon 835 with sufficient active/liquid cooling to overclock CPU, GPU, memory.
I am going to CES 2018 to see and test the latest from Qualcomm and others.
My application is the **[TeleCuris TeleHealth IoMT Kit](http://telecuris.com/products)** is a TeleHealth [Internet of Medical Things](https://www.microsoft.com/en-us/internet-of-things/healthcare?&WT.srch=1&WT.mc_id=AID559320_SEM_jI1mhjxr&ef_id=WgR51QAAAIxEmmpZ:20171127212524:s) (IoMT) Remote Patient Diagnoses Kit consist of various remote vital sign diagnostic devices.
As soon as MS fixes the broken installation of **Azure Machine Learning Workbench on Windows** I'll blog about it at **[Ivan-Farkas.com](http://ivan-farkas.com/)**
See the the detailed error in my comment (TeleCuris) at this [link](https://docs.microsoft.com/en-us/azure/machine-learning/preview/quickstart-installation).
```
Installation of Azure Machine Learning Workbench on Windows fails!
It can't install ca-certificates 2017.7.27.
```

I am looking forward to seeing your project results and compare notes. Good luck.

lovettchris, 2017-11-27:
Awesome applications.  I like Eric-tang's description of the difference between Azure IoT Edge and ELL.  I think of the Azure service as a horizontal solution that pulls everything together and I think of ELL as a building block that could be deployed on IoT devices used in such an end-to-end system.

We've definitely run ELL on Dragonboad 410c, so the 820c should be fine.  The 410 is twice as fast as the Raspberry Pi.  Love to see how the 820 performs, so I look forward to your next blog!

IvanFarkas, 2017-11-27:
I'll do a blog next week about the **Dragonboad 410c** and compare it to a **SnapDragon 820** Dev Kit as soon as I get one.

IvanFarkas, 2017-11-28:
@lovettchris Chris. What drone platform would you use in the next 6-12 month for the open source PX4 flight stack that I can integrate with **ELL**?

- [Pixhawk](https://pixhawk.org/)
- [Qualcomm® Snapdragon™ Flight™ Development Kit](https://www.intrinsyc.com/vertical-development-platforms/qualcomm-snapdragon-flight/)
- [Intel Aero ready to fly drone](https://software.intel.com/en-us/aero)

lovettchris, 2017-11-28:
Hey, ELL and PX4, my favorite combination :-)   Both the Qualcomm Flight kit and the intel Aero are using the PX4 stack, so either kit is a good choice.  See also my [MavLinkCom](https://github.com/Microsoft/AirSim/tree/master/MavLinkCom) library which is part of AirSim.  The idea with Airsim is you can test our vision algorithms in a simulator before trying them on a drone :-)

IvanFarkas, 2017-11-28:
I'll sure do Chris. 2018 is gonna be a great year for **IoT**, **Machine Learning**, and **intelligent drones**!

lovettchris, 2017-11-28:
Agreed!  I feel like there is more opportunity to do really amazing stuff with computing right now than there ever has been.

IvanFarkas, 2017-11-29:
@lovettchris Chris. How do I get in touch with you offline, please?

closed 2017-11-29 by IvanFarkas