microsoft/ELL #115: a trouble of darknet to ell

Tryljs, 2017-11-21:
Dear team
I call the function to get predictor from 'model.ell' as follows:

int darknet_to_ell_test()
{
	IplImage *frame;
	CvCapture *capture = cvCreateCameraCapture(0);
	// add arguments to the command line parser
	common::ParsedDataLoadArguments dataLoadArguments;
	common::ParsedDataSaveArguments dataSaveArguments;
	common::ParsedMapLoadArguments mapLoadArguments;

	// load map
	mapLoadArguments.inputMapFilename = "F:\\darnet_to_ell_map\\model.ell";
	auto map = common::LoadMap(mapLoadArguments);

	// get predictor
	auto model = map.GetModel();
	predictors::NeuralNetworkPredictor<float> predictor;
	auto iterNode = model.GetNodeIterator();
	while (iterNode.IsValid())
	{
		auto node = iterNode.Get();
		auto nodeType = node->GetRuntimeTypeName();
		if (nodeType == "NeuralNetworkPredictorNode<float>")
		{
			const ell::nodes::NeuralNetworkPredictorNode<float>* predictorNode = dynamic_cast<const ell::nodes::NeuralNetworkPredictorNode<float>*>(node);
			predictor = predictorNode->GetPredictor();
		}	
		iterNode.Next();
	}

	cvNamedWindow("Display");
	while (1)
	`{`
		// frame = cvQueryFrame(capture);

		frame = cvLoadImage("F:\\darnet_to_ell_map\\mouse.png", 1);
		cvShowImage("Display", frame);

		std::vector<double> data = prepare_image_for_predictor(frame);
		data::FloatDataVector iputData(data);

		clock_t start = clock();
		auto predictions = predictor.Predict(iputData);
		clock_t end = clock();
		double need_time = (double)((end - start) / CLOCKS_PER_SEC);
		cout << need_time << "s :";

		for (int i = 0; i < predictions.size(); i++)
		{
			if (predictions.at(i) > 0.25)
		            cout << "  index="<< i <<  "  val="<< predictions.at(i) << endl;	
		}
		char c = cvWaitKey(33);
		if (c == 27)  break;
	}
       cvDestroyAllWindows();
      cvReleaseCapture(&capture);
      return 0;
}

The function of the ‘prepare_image_for_predictor(frame)’ function is the same as modelHelper's function.

Use the above function to predict a target,  such as mouse. But it takes about 3 minutes to call the function 'LoadMap()'. And then, get the predictor from model takes about 2 minutes. Lastly, call the function 'predictor.Predict(iputData)' takes about 9 seconds and the values of all 'predictions' were less than 0.25. It's much slower than ‘darknetDemo.py’ demo. I used ‘darknetDemo.py’ demo to predict 'mouse.png' very quickly, it takes about 0.6 seconds and the score was 0.99.

What's wrong with my program? And how do I modify it?

lovettchris, 2017-11-21:
Are you running debug bits perhaps?  The Release build is much quicker.  Also when I switched your code to use the [TutorialHelpers ](https://github.com/Microsoft/ELL/tree/master/docs/tutorials/Getting-started-with-image-classification-in-cpp) for doing center cropping of the image I was able to get your code to produce correct results.  Release bits detected [this elephant](https://www.bing.com/images/search?view=detailV2&ccid=%2BNZ5o2aT&id=2B78DE81C71D8800498545755EB766BDC002C055&thid=OIP.-NZ5o2aTrJ9SOjtMv4ZxewEsDU&mediaurl=http%3A%2F%2Fimagenes.4ever.eu%2Fdata%2Fdownload%2Fanimales%2Fsalvajes%2Felefante%2C-verde-191999.jpg&exph=567&expw=800&q=elephant&ajaxhist=0&pivotparams=insightsToken%3Dccid_F58N%252B6nv*mid_67956C86297BF8841FD371724A23338416F63F9F*simid_608030971466285795*thid_OIP.F58N-6nvsqKklWm3dCYXEAEjCk&iss=VSI) in about 0.9 seconds on my HP Z840 desktop machine.

lovettchris, 2017-11-21:
Alternatively, if you compile your darknet ELL model instead using the technique described in [this tutorial ](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-in-cpp/) then you will not only find it is much easier to compile a C++ application around that it also runs a lot faster.

Simply include the compiled "darknet.i.h" and the "darknet.obj" into your project and call the predict darknet_Predict function.  This version runs predict in about 23 milliseconds.

```cpp
void test_compiled_ell_model() 
{
    cv::Mat frame;

    std::vector<float> predictions;
    predictions.resize(1000);

    cvNamedWindow("Display");
    while (1)
    {
        // frame = cvQueryFrame(capture);

        frame = cv::imread("d:\\temp\\elephant.png");
        cv::imshow("ELL model", frame);

        std::vector<float> data = TutorialHelpers::PrepareImageForModel(frame, 224, 224, true);

        clock_t start = clock();
        darknet_Predict(data, predictions);
        clock_t end = clock();
        double need_time = (double)((end - start) * 1000 / CLOCKS_PER_SEC);
        cout << need_time << "ms :";

        for (int i = 0; i < predictions.size(); i++)
        {
            if (predictions.at(i) > 0.25)
                cout << "  index=" << i << "  val=" << predictions.at(i) << endl;
        }
        char c = cvWaitKey(33);
        if (c == 27)  break;
    }
}
```

Tryljs, 2017-11-23:
Thanks too much! I've done to recognition the target  with your demo. 
 And another question, does the 'darknet_Predict' function can return the location information of the target, such as top point, size and so on ?

lovettchris, 2017-11-23:
You're welcome, glad to hear it works better now.  ELL does not support region detection models, yet....

Tryljs, 2017-11-24:
Now, I want to run ELL on Cortex-A9. But it just support target platforms are pi3, aarch64 and host, none of them are ARMv7 framework. And I find the cross compiling of 'arm-poky-linux-gnueabi-gcc' coundn't to compile ELL, because the std of 'arm-poky-linux-gnueabi-gcc' is not supported. How can I do for ELL to make it run on Cortex-A9?

lovettchris, 2017-12-02:
moved to #120

closed 2017-12-02 by lovettchris