microsoft/ELL #116: when does ELL support micro:bit

huoyunjie, 2017-11-24:
Hi, all,
    This project is very cool. I develop some products with ARM Cortex-M processor. I want to know when we can run ELL in  micro:bit.
    Thanks for your hard work on this project!

Tryljs, 2017-12-12:
我看了下生成的文件  应该是支持ARM端的

huoyunjie, 2017-12-12:
多谢！我还没跑过，一直没有看到在micro:bit上运行的例子！

Tryljs, 2017-12-12:
这个我是猜的  我也没有实现

lovettchris, 2018-04-24:
We have not done it yet, and the nrf51822 chip on the microbit is very resource constrained (256kb flash and 32 kb RAM), so it would only be able to run a very small model, perhaps a ProtoNN based model might work.  I have used ELL to compile a model for custom ARM cortex-M4 using the following compile.exe command line:
```
ELL\build\bin\release\compile … --target custom --cpu cortex-m0 --triple armv6m-unknown-none-eabi --features +armv6-m,+v6m --numBits 32 

```
But ELL models typically use a lot of floating point, so I would recommend instead you choose a board that has a Cortex-M4f processor at least, like the newer nRF52840 from Nordic.  The following command line builds Cortex-M4F code, making use of the vectorized floating point instructions:

```
ELL\build\bin\release\compile … --target custom --numBits 32 --cpu cortex-m4 --triple armv6m-gnueabi --features +vfp4-sp-d16,+soft-float
```

lovettchris, 2018-11-08:
See also [Building an ELL model for ARM Cortex-M4](https://github.com/Microsoft/ELL/wiki/Building-an-ELL-model-for-ARM-Cortex-M4)

closed 2018-11-08 by lovettchris