microsoft/ELL #117: CMake cannot locate OpenBLAS

IvanFarkas, 2017-11-24:
Followed build instructions.

```shell
setx -m ELL_Src C:\ML\EEL\ELL
setx -m ELL_bin %ELL_Src%\build\bin\Release
setx -m OpenBLASLibs %ELL_Src%\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\bin
setx -m PATH %PATH%;%OpenBLASLibs%

cd %ELL_Src%
nuget.exe restore external/packages.config -PackagesDirectory external

conda create -n py36 numpy python=3.6
activate py36

mkdir build
cd build
cmake -G "Visual Studio 15 2017 Win64" ..
```

**Expected Result** (OpenBLAS section only)

```shell
-- Processor family: 6, model: 158
-- Using OpenBLAS compiled for haswell
-- Using BLAS include path: C:/ML/EEL/ELL/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/include
-- Using BLAS library: C:/ML/EEL/ELL/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/lib/libopenblas.dll.a
-- Using BLAS DLLs: libopenblas.dll;libgcc_s_seh-1.dll;libgfortran-3.dll;libquadmath-0.dll
```

**Actual Result** (OpenBLAS section only)

```shell
-- Blas libraries: C:/ProgramData/Miniconda3/Library/lib/mkl_intel_lp64_dll.lib;C:/ProgramData/Miniconda3/Library/lib/libiomp5md.lib;C:/ProgramData/Miniconda3/Library/lib/mkl_intel_thread_dll.lib;C:/ProgramData/Miniconda3/Library/lib/mkl_core_dll.lib
-- Blas linker flags:
-- Blas include directories: C:/ML/EEL/ELL/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/include
-- Using BLAS include path: C:/ML/EEL/ELL/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/include
-- Using BLAS library: C:/ProgramData/Miniconda3/Library/lib/mkl_intel_lp64_dll.lib;C:/ProgramData/Miniconda3/Library/lib/libiomp5md.lib;C:/ProgramData/Miniconda3/Library/lib/mkl_intel_thread_dll.lib;C:/ProgramData/Miniconda3/Library/lib/mkl_core_dll.lib
-- Using BLAS DLLs:
```

The following modification to \CMake\OpenBLASSetup.cmake solved the issue

**Before**
```cmake
find_package(BLAS QUIET)
```

**After**
```cmake
if(NOT WIN32)
    find_package(BLAS QUIET)
endif()
```

kernhanda, 2017-11-28:
Thanks @IvanFarkas. We'll work on cleaning up the CMake configuration output. Please let us know if you were unable to build.

IvanFarkas, 2017-11-29:
ELL built successfully.

closed 2017-11-29 by IvanFarkas