microsoft/ELL #118: CMake cannot locate LLVM

IvanFarkas, 2017-11-24:
Followed build instructions.

```shell
setx -m ELL_Src C:\ML\EEL\ELL
setx -m ELL_bin %ELL_Src%\build\bin\Release
setx -m OpenBLASLibs %ELL_Src%\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\bin
setx -m PATH %PATH%;%OpenBLASLibs%

cd %ELL_Src%
nuget.exe restore external/packages.config -PackagesDirectory external

conda create -n py36 numpy python=3.6
activate py36

mkdir build
cd build
cmake -G "Visual Studio 15 2017 Win64" ..
```

**Error** (LLVM section only)

```shell
CMake Warning at CMake/LLVMSetup.cmake:48 (find_package):
  By not providing "FindLLVM.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LLVM", but
  CMake did not find one.

  Could not find a package configuration file provided by "LLVM" (requested
  version 3.9) with any of the following names:

    LLVMConfig.cmake
    llvm-config.cmake

  Add the installation prefix of "LLVM" to CMAKE_PREFIX_PATH or set
  "LLVM_DIR" to a directory containing one of the above files.  If "LLVM"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:69 (include)
```

**CMake output**
[2.1 - CMake output.txt](https://github.com/Microsoft/ELL/files/1502336/2.1.-.CMake.output.txt)

kernhanda, 2017-11-28:
This is just a warning and it can be safely ignored. We're working to get rid of it.

IvanFarkas, 2017-11-29:
ELL built successfully.

closed 2017-11-29 by IvanFarkas