microsoft/ELL #121: ELL build fails on WSL

IvanFarkas, 2017-11-29:
ELL build fails on WSL. See CMake and Make outputs attached.
WSL Ubuntu 16.04.3 LTS has the latest updates.

`
[ 61%] Linking CXX executable common_test
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Copy(int, float const*, int, float*, int)':
BlasWrapper.cpp:(.text+0x99): undefined reference to `cblas_scopy'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Copy(int, double const*, int, double*, int)':
BlasWrapper.cpp:(.text+0xd1): undefined reference to `cblas_dcopy'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Asum(int, float const*, int)':
BlasWrapper.cpp:(.text+0xfa): undefined reference to `cblas_sasum'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Asum(int, double const*, int)':
BlasWrapper.cpp:(.text+0x122): undefined reference to `cblas_dasum'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Nrm2(int, float const*, int)':
BlasWrapper.cpp:(.text+0x158): undefined reference to `cblas_snrm2'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Nrm2(int, double const*, int)':
BlasWrapper.cpp:(.text+0x180): undefined reference to `cblas_dnrm2'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Scal(int, float, float*, int)':
BlasWrapper.cpp:(.text+0x1c3): undefined reference to `cblas_sscal'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Scal(int, double, double*, int)':
BlasWrapper.cpp:(.text+0x1fb): undefined reference to `cblas_dscal'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Axpy(int, float, float const*, int, float*, int)':
BlasWrapper.cpp:(.text+0x241): undefined reference to `cblas_saxpy'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Axpy(int, double, double const*, int, double*, int)':
BlasWrapper.cpp:(.text+0x289): undefined reference to `cblas_daxpy'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Dot(int, float const*, int, float const*, int)':
BlasWrapper.cpp:(.text+0x2c1): undefined reference to `cblas_sdot'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Dot(int, double const*, int, double const*, int)':
BlasWrapper.cpp:(.text+0x2f8): undefined reference to `cblas_ddot'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Ger(ell::math::MatrixLayout, int, int, float, float const*, int, float const*, int, float*, int)':
BlasWrapper.cpp:(.text+0x36d): undefined reference to `cblas_sger'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Ger(ell::math::MatrixLayout, int, int, double, double const*, int, double const*, int, double*, int)':
BlasWrapper.cpp:(.text+0x3db): undefined reference to `cblas_dger'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemv(ell::math::MatrixLayout, ell::math::MatrixTranspose, int, int, float, float const*, int, float const*, int, float, float*, int)':
BlasWrapper.cpp:(.text+0x465): undefined reference to `cblas_sgemv'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemv(ell::math::MatrixLayout, ell::math::MatrixTranspose, int, int, double, double const*, int, double const*, int, double, double*, int)':
BlasWrapper.cpp:(.text+0x4f6): undefined reference to `cblas_dgemv'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemm(ell::math::MatrixLayout, ell::math::MatrixTranspose, ell::math::MatrixTranspose, int, int, int, float, float const*, int, float const*, int, float, float*, int)':
BlasWrapper.cpp:(.text+0x596): undefined reference to `cblas_sgemm'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemm(ell::math::MatrixLayout, ell::math::MatrixTranspose, ell::math::MatrixTranspose, int, int, int, double, double const*, int, double const*, int, double, double*, int)':
BlasWrapper.cpp:(.text+0x63c): undefined reference to `cblas_dgemm'
collect2: error: ld returned 1 exit status
libraries/common/CMakeFiles/common_test.dir/build.make:190: recipe for target 'libraries/common/common_test' failed
make[2]: *** [libraries/common/common_test] Error 1
CMakeFiles/Makefile2:264: recipe for target 'libraries/common/CMakeFiles/common_test.dir/all' failed
make[1]: *** [libraries/common/CMakeFiles/common_test.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
`

[CMake G output.txt](https://github.com/Microsoft/ELL/files/1514508/CMake.G.output.txt)
[Make build output.txt](https://github.com/Microsoft/ELL/files/1514510/Make.build.output.txt)

kernhanda, 2017-11-30:
Did you run `sudo apt-get install -y libopenblas-dev` per the [Ubuntu install instructions](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md)?

IvanFarkas, 2017-11-30:
Twice

kernhanda, 2017-11-30:
Can you try passing `-DBLA_VENDOR=OpenBLAS` and then try building? Please make sure to delete your `build` folder and create it again.

IvanFarkas, 2017-11-30:
That worked. Compiled successfully. Please update the documents. Thx for your help.

closed 2017-11-30 by IvanFarkas

lovettchris, 2017-11-30:
Weird, I'm not getting this also using WSL, there must be something different about our openblas setups...

[image]

```
chris@clovett14:/mnt/d/temp/ELL$ ls -al /usr/lib/libopen*
lrwxrwxrwx 1 root root       22 Apr 19  2016 /usr/lib/libopenblas.a -> libopenblasp-r0.2.18.a
-rw-r--r-- 1 root root 50859686 Apr 19  2016 /usr/lib/libopenblasp-r0.2.18.a
-rw-r--r-- 1 root root 31962416 Apr 19  2016 /usr/lib/libopenblasp-r0.2.18.so
lrwxrwxrwx 1 root root       23 Apr 19  2016 /usr/lib/libopenblas.so -> libopenblasp-r0.2.18.so
lrwxrwxrwx 1 root root       23 Apr 19  2016 /usr/lib/libopenblas.so.0 -> libopenblasp-r0.2.18.so
lrwxrwxrwx 1 root root       16 Feb 25  2016 /usr/lib/libopen-pal.so -> libopen-pal.so.4
lrwxrwxrwx 1 root root       21 Feb 25  2016 /usr/lib/libopen-pal.so.13 -> libopen-pal.so.13.0.2
lrwxrwxrwx 1 root root       33 Feb 25  2016 /usr/lib/libopen-pal.so.13.0.2 -> openmpi/lib/libopen-pal.so.13.0.2
lrwxrwxrwx 1 root root       16 Feb 25  2016 /usr/lib/libopen-rte.so -> libopen-rte.so.4
lrwxrwxrwx 1 root root       21 Feb 25  2016 /usr/lib/libopen-rte.so.12 -> libopen-rte.so.12.0.2
lrwxrwxrwx 1 root root       33 Feb 25  2016 /usr/lib/libopen-rte.so.12.0.2 -> openmpi/lib/libopen-rte.so.12.0.2```

IvanFarkas, 2017-11-30:
What about after
```shell
sudo apt update && sudo apt upgrade
```