microsoft/ELL #124: Erro:Building the ELL on Ubuntu Linux

Tryljs, 2017-12-29:
Building message:
......
[ 55%] Building CXX object libraries/common/CMakeFiles/common.dir/src/LoadModel.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
libraries/common/CMakeFiles/common.dir/build.make:158: recipe for target 'libraries/common/CMakeFiles/common.dir/src/LoadModel.cpp.o' failed
make[2]: *** [libraries/common/CMakeFiles/common.dir/src/LoadModel.cpp.o] Error 4
CMakeFiles/Makefile2:216: recipe for target 'libraries/common/CMakeFiles/common.dir/all' failed
make[1]: *** [libraries/common/CMakeFiles/common.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

kernhanda, 2017-12-29:
This looks to be a fault with the compiler. It should never be killed when compiling a file. Please file a bug report for your compiler and/or upgrade your compiler.

For the sake of curiousity, what's the version of your compiler?

Tryljs, 2018-01-02:
My compiler's version is 5.4.0.20160609

Tryljs, 2018-01-02:
I upgrade my compiler version with 6.3.0. But it doesn't work.

kernhanda, 2018-01-04:
Does it result in the same error? Can you please run the following (feel free to adjust the `cmake` commands if you were doing something different, but please call it out) in the ELL repository

```shell
git clean -fdx
mkdir build
cd build
cmake ..
cmake --build .
```
and share the output?

Tryljs, 2018-01-08:
This problem had been solved.

kernhanda, 2018-01-08:
Thanks. What was the problem?

Tryljs, 2018-01-08:
Lack of caching

closed 2018-01-08 by kernhanda

JayjieL, 2018-06-17:
so how to solve this problem....