microsoft/ELL #125: Trouble: complier darknet project

Tryljs, 2018-01-08:
I used wrap.py script to process the darknet model, and then  I got a darknet.obj file. When I used it to complier a demo, but failed. Error message as following:

darknet.obj : error LNK2019: in function 'darknet_Predict', undefined reference to 'darknet_InputCallback';
darknet.obj : error LNK2019: in function 'darknet_Predict', undefined reference to 'cblas_sgemm';
darknet.obj : error LNK2019: in function 'darknet_Predict', undefined reference to 'darknet_OutputCallback';

Tryljs, 2018-01-08:
In the predict function, I found its parameters are "double", "float*", how to input my picture to predict the result?

lisaong, 2018-01-09:
This is now fixed with release v2.1.5

closed 2018-01-09 by lisaong