microsoft/ELL #126: Darknet to ell error

Tryljs, 2018-01-09:
I updated the ELL code and runned  the 'rebuild.cmd' script to finish compiler the ELL project. Then I used my script to run 'darknet_import.py' script, but failed. My script content was 'python ../ELL_20180109/tools/importers/darknet/darknet_import.py darknet.cfg darknet.weights', error message as following:

E:\darknet_to_ell>python ../ELL_20180109/tools/importers/darknet/darknet_import.
py darknet.cfg darknet.weights'
Traceback (most recent call last):
  File "E:\ELL_20180109\build\interfaces\python\package\ell\ell_py.py", line 14,
 in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\import
lib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 948, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'ell._ell_py'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "../ELL_20180109/tools/importers/darknet/darknet_import.py", line 17, in
<module>
    import find_ell
  File "E:\ELL_20180109\tools\importers\darknet\../../utilities/pythonlibs\find_
ell.py", line 61, in <module>
    import ell
  File "E:\ELL_20180109\build\interfaces\python\package\ell\__init__.py", line 2
6, in <module>
    from . import data
  File "E:\ELL_20180109\build\interfaces\python\package\ell\data\__init__.py", l
ine 9, in <module>
    from ..ell_py import AutoDataVector, \
  File "E:\ELL_20180109\build\interfaces\python\package\ell\ell_py.py", line 17,
 in <module>
    _ell_py = swig_import_helper()
  File "E:\ELL_20180109\build\interfaces\python\package\ell\ell_py.py", line 16,
 in swig_import_helper
    return importlib.import_module('_ell_py')
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\import
lib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: Unable to find the specified module.

E:\darknet_to_ell>pause

lisaong, 2018-01-09:
Are you running from within a conda environment?  From the spew, looks like you are running the system-installed python.

```
activate py36
python .../darknet_import.py darknet.cfg darknet.weights
```

Tryljs, 2018-01-10:
I don't think. Before I update the code, my script runs in the Windows and it can successfully execute the 'darknet_import.py'. I executed the ELL code was 2017-12-28. The running message as following:

E:\darknet_to_ell>python ../ELL_20171228/tools/importers/darknet/darknet_import.
py darknet.cfg darknet.weights'
Convolution : 224x224x3 -> 224x224x16 | input padding 1 output padding 0
......
Saving model file: 'darknet.ell'

E:\darknet_to_ell>pause

lisaong, 2018-01-10:
I see, can you compare what's under these folders? 

dir /s E:\ELL_20171228\build\interfaces\python\package\ell
dir /s E:\ELL_20180109\build\interfaces\python\package\ell

julian0001, 2018-06-17:
Hallo @Tryljs have you solved your issue with the ModuleNotFoundError of ell._ell_py? Because I have the same issue now with CNTK.

Tryljs, 2018-06-27:
Not

psiphi75, 2019-01-17:
I had a similar issue under Linux and it appears that I did not compile ELL in the conda environment.

Hence running the following before compiling ELL:
```sh
source activate py36
```

And making sure that before running any ELL commands you do the same.

lovettchris, 2019-01-17:
and if you didn't compile ELL in the conda environment, you will need to run "rebuild.cmd" to ensure you delete the "build" folder, otherwise cmake will not update the make files needed to build the ELL python modules.

lovettchris, 2019-04-29:
closing due to inactivity, please reopen if you are still waiting for an update on this.

closed 2019-04-29 by lovettchris