microsoft/ELL #127: Error when trying to compile tutorial.cpp with make

jesuspicazo, 2018-01-15:
Although it was working OK for the last weeks, last Friday tried to compile the ELL classification tutorial for c++ and obtained the following error:
/home/icai23/3/tutorial.cpp:83:41: error: no matching function for call to ‘model_Predict(std::vector<float>&, std::vector<float>&)’
         model_Predict(input, predictions);
                                         ^
In file included from /home/icai23/3/tutorial.cpp:16:0:
/home/icai23/3/model/model.i.h:42:6: note: candidate: void model_Predict(double, float*)
 void model_Predict(double, float*);
      ^
/home/icai23/3/model/model.i.h:42:6: note:   no known conversion for argument 1 from ‘std::vector<float>’ to ‘double’
/home/icai23/3/model/model.i.h:71:6: note: candidate: void model_Predict(double, std::vector<float>&)
 void model_Predict(double input, std::vector<float>& output)
      ^
/home/icai23/3/model/model.i.h:71:6: note:   no known conversion for argument 1 from ‘std::vector<float>’ to ‘double’
CMakeFiles/tutorial.dir/build.make:62: recipe for target 'CMakeFiles/tutorial.dir/tutorial.cpp.o' failed
make[2]: *** [CMakeFiles/tutorial.dir/tutorial.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/tutorial.dir/all' failed
make[1]: *** [CMakeFiles/tutorial.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I don't know the reason for this as it was working OK before.

lisaong, 2018-01-16:
Thanks for reporting this.  There has been an update to the C++ interface but the tutorial has not yet caught up. We'll be fixing this hopefully within the next couple days.

jesuspicazo, 2018-01-16:
Hi Lisa. Thanks to you for your replay. Is there any quick workaround to solve this until you have fixed the tutorial? I would be very thankful for that.

Thank you very much.

Best regards.

lovettchris, 2018-01-18:
Here's a version of the tutorial that works with the new callback interface:

```cpp
////////////////////////////////////////////////////////////////////////////////////////////////////
//
//  Project:  Embedded Learning Library (ELL)
//  File:     tutorial.cpp
//  Authors:  Byron Changuion
//
////////////////////////////////////////////////////////////////////////////////////////////////////

#include <stdio.h>
#include <fstream>
#include <sstream>
#include <chrono>
#include <opencv2/opencv.hpp>

// Include the model interface file for the compiled ELL model
#include "model.h"

// Include helper functions
#include "tutorialHelpers.h"

// Read an image from the camera
static cv::Mat GetImageFromCamera(cv::VideoCapture& camera)
{
    cv::Mat frame;
    camera >> frame;
    return frame;
}

// Read an image from a file
static cv::Mat GetImageFromFile(const std::string& filename)
{
    return cv::imread(filename);
}

// Read a file of strings
static std::vector<std::string> ReadLinesFromFile(const std::string& filename)
{
    std::vector<std::string> lines;
    std::string line;

    std::ifstream file(filename);

    while (std::getline(file, line))
    {
        if (line.length() > 0) lines.emplace_back(line);
    }

    return lines;
}

struct GlobalContext {    
  cv::VideoCapture camera;
  TensorShape inputShape;
  TensorShape outputShape;
  size_t inputSize;
  std::vector<float> input;
  cv::Mat image;
  size_t outputSize;
  std::vector<float> predictions;
};

GlobalContext context;

int8_t model_InputCallback(float* buffer)
{
    // Get an image from the camera. (Alternatively, call GetImageFromFile to read from file)
  cv::Mat image = GetImageFromCamera(context.camera);
  context.image = image;

  // Prepare an image for processing
  // - Resize and center-crop to the required width and height while preserving aspect ratio.
  // - OpenCV gives the image in BGR order. If needed, re-order the channels to RGB.
  // - Convert the OpenCV result to a std::vector<float>
  std::vector<float> input = tutorialHelpers::PrepareImageForModel(image, context.inputShape.columns, context.inputShape.rows);

  size_t size = input.size();
  assert(size == context.inputSize);
  ::memcpy(buffer, &input[0], context.inputSize * sizeof(float));

  return true;
}

void model_OutputCallback(float* buffer)
{
    context.predictions = std::vector<float>(&buffer[0], &buffer[context.outputSize]);
}

int main(int argc, char** argv)
{
    context.camera = cv::VideoCapture(1);

    // Read the category names
    auto categories = ReadLinesFromFile("categories.txt");

    // Get the model's input shape. We will use this information later to resize images appropriately.

    // Get the model's input shape. We will use this information later to resize images appropriately.
    TensorShape inputShape;
    model_GetInputShape(0, &inputShape);
    context.inputShape = inputShape;
    context.inputSize = model_GetInputSize();
    printf("Input shape=[%d,%d,%d]\n", inputShape.rows, inputShape.columns, inputShape.channels);
   
    TensorShape outputShape;
    model_GetOutputShape(0, &outputShape);
    context.outputShape = outputShape;
    printf("Output shape=[%d,%d,%d]\n", outputShape.rows, outputShape.columns, outputShape.channels);

    // Create a vector to hold the model's output predictions

    context.outputSize = model_GetOutputSize();
    context.input.resize(context.inputSize);

    if (categories.size() != context.outputSize) 
    {
        printf("categories.txt does not contain %d categories\n", context.outputSize);
        return 1;
    }
    auto start = std::chrono::steady_clock::now();

    // Declare a variable to hold the prediction times
    std::vector<double>  predictionTimes;
    double meanTimeToPredict = 0.0;

    while ((cv::waitKey(1) & 0xFF) == 0xFF)
    {
        // Send the image to the compiled model and fill the predictions vector with scores, measure how long it takes
        auto now = std::chrono::steady_clock::now();
        double ticksFromStart = std::chrono::duration<double>(now - start).count();
        double ticks = model_GetTicksUntilNextInterval(ticksFromStart);
        // bugbug: what is the second parameter?  Why do I need to provide a buffer here, the debugger shows it is not
        // the same one passed to model_InputCallback...
        model_Predict(ticks, context.input.data());
        auto end = std::chrono::steady_clock::now();

        if (context.predictions.size() > 0)
        {
            // Get the value of the top 5 predictions
            // todo: for multithreaded models, this will need a lock to safely access predictions
            auto top5 = tutorialHelpers::GetTopN(context.predictions, 5);

            // Generate header text that represents the top5 predictions
            std::stringstream headerText;
            for (auto element : top5)
            {
                headerText << "(" << std::floor(element.second * 100.0) << "%) " << categories[element.first] << "  ";
            }
            cv::Mat image = context.image;
            tutorialHelpers::DrawHeader(image, headerText.str());

            // Generate footer text that represents the mean evaluation time
            std::stringstream footerText;
            meanTimeToPredict = std::floor(tutorialHelpers::GetMeanDuration(predictionTimes, std::chrono::duration<double>(end - start).count()) * 1000);
            footerText << meanTimeToPredict << "ms/frame";
            tutorialHelpers::DrawFooter(image, footerText.str());

            // Display the image
            cv::imshow("ELL model", image);
        }
    }

    std::cout << "Mean prediction time: " << meanTimeToPredict << "ms/frame" << std::endl;

    return 0;
}
```

closed 2018-01-23 by lovettchris