microsoft/ELL #128: ELL compile error on Mac: EmitterTypes.h not found

kvijaykiran, 2018-01-18:
Hi,

I end up with the following message as I issue make after all dependencies were satisfied. I see this error on two different Macs, one running Sierra and the other running High Sierra:

`/Users/xxxxx/Development/EmbeddedLearningLibrary/ELL/libraries/model/include/IRModelProfiler.h:15:10: fatal error: 
      'EmitterTypes.h' file not found
#include "EmitterTypes.h"
         ^~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o] Error 1
make[1]: *** [libraries/model/CMakeFiles/model.dir/all] Error 2
make: *** [all] Error 2
(py36) bash-3.2$ 
`

The CMakeLists files look correct to me, but I am probably missing something. Any thoughts?

- Vijay.

kernhanda, 2018-01-20:
@kvijaykiran - Perhaps the instructions at https://github.com/Microsoft/ELL/issues/1#issuecomment-359204598 will prove to be helpful for you

lovettchris, 2019-04-29:
closing due to inactivity, please reopen if you are still waiting for an update on this.

closed 2019-04-29 by lovettchris