microsoft/ELL #129: retargetTrainer generates old style ell models

m-zara, 2018-01-24:
I have downloaded today an updated model from ELL-models repository. retargetTrainer generates old style model, although I give this new style trained model as input.

lisaong, 2018-01-26:
Hi, can you describe what "old style" model means?

m-zara, 2018-01-26:
Your new ELL model generator from CNTK puts these layers at the beginning:

"nodes": [
    {
      "_type": "InputNode<double>",
      "_version": "2",
      "id":       "1012",
      "size": 1,
      "shape": [1, 1, 1]
    }, 
    {
      "_type": "ClockNode",
      "_version": "0",
      "id":       "1013",
      "input": {
        "_type": "InputPort<double>",
        "_version": "0",
        "nodeId":         "1013",
        "name": "input",
        "type": 2,
        "input":         "1012.output[0]"
      },
      "output": {
        "_type": "OutputPort<double>",
        "_version": "0",
        "nodeId":         "1013",
        "name": "output",
        "type": 2,
        "size": 2
      },
      "interval": 0,
      "lagThreshold": 0,
      "lagNotificationFunctionName": "LagNotification"
    }, 
    {
      "_type": "SourceNode<float>",
      "_version": "4",
      "id":       "1014",
      "input": {
        "_type": "InputPort<double>",
        "_version": "0",
        "nodeId":         "1014",
        "name": "input",
        "type": 2,
        "input":         "1013.output[0:2]"
      },
      "output": {
        "_type": "OutputPort<float>",
        "_version": "0",
        "nodeId":         "1014",
        "name": "output",
        "type": 1,
        "size": 12288
      },
      "sourceFunctionName": "InputCallback",
      "shape": [64, 64, 3]
    }, 
    {
      "_type": "NeuralNetworkPredictorNode<float>",
      "_version": "0",
      "id":       "1015",
      "input": {
        "_type": "InputPort<float>",
        "_version": "0",
        "nodeId":         "1015",
        "name": "input",
        "type": 1,
        "input":         "1014.output[0:12288]"
      },

But retargetTrainer has not been updated yet, this function does not put above layers at the beginning, just these ones:

"nodes": [
    {
      "_type": "InputNode<float>",
      "_version": "2",
      "id":       "1006",
      "size": 12288,
      "shape": [64, 64, 3]
    }, 
    {
      "_type": "NeuralNetworkPredictorNode<float>",
      "_version": "0",
      "id":       "1007",
      "input": {
        "_type": "InputPort<float>",
        "_version": "0",
        "nodeId":         "1007",
        "name": "input",
        "type": 1,
        "input":         "1006.output[0:12288]"
      },

lisaong, 2018-01-30:
Thanks for the clarification, will take a look and get back shortly

lisaong, 2018-02-07:
Hi! This is now fixed with v2.2.0.

closed 2018-02-07 by m-zara