microsoft/ELL #130: Fatal Python occurred when I call warp.py

YanMinge, 2018-01-25:
My development environment is：
MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)
macOS High Sierra 10.13.2

I followed the instructions [https://github.com/Microsoft/ELL/blob/master/INSTALL-Mac.md](https://github.com/Microsoft/ELL/blob/master/INSTALL-Mac.md) complete the construction of the environment， No error message for `make` and `make _ELL_python `

but when I follow this instruction to do the sample application
[https://github.com/Microsoft/ELL/blob/master/docs/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/index.md(https://github.com/Microsoft/ELL/blob/master/docs/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/index.md)
The following error has appeared：
```
(py36) yanmingedeMacBook-Pro:1_RPI_Test yanminge$ python ~/project/ELL/tools/wrap/wrap.py model.ell -lang python -target host
Fatal Python error: PyThreadState_Get: no current thread

Abort trap: 6
```

and my python environment is 
```
(py36) yanmingedeMacBook-Pro:1_RPI_Test yanminge$ python --version
Python 3.6.4 :: Anaconda, Inc.
(py36) yanmingedeMacBook-Pro:1_RPI_Test yanminge$ which python
/Users/yanminge/miniconda3/envs/py36/bin/python
(py36) yanmingedeMacBook-Pro:1_RPI_Test yanminge$ 
```
Can anyone help me?

lisaong, 2018-01-26:
Hi, as you may have found online, this error happens on Macs when various versions of python conflict. 

As this is a setup problem with the Mac environment, here are some references that may help:

https://github.com/snap-stanford/snap/issues/88

...which points to this more detailed troubleshooting guide for something else, but may help you here:

https://docs.google.com/document/d/1ODvtQsJgiTOJH4-qBqwa6eAvoNJoiALqnk2pxFf1IzY/edit

YanMinge, 2018-01-28:
I also searched for the same content on google，Because I used to refer to this document to create the environment [https://github.com/Microsoft/ELL/blob/master/INSTALL-Mac.md](https://github.com/Microsoft/ELL/blob/master/INSTALL-Mac.md)，So I am using a virtual environment，and you can see my python environment returned the expected result.

YanMinge, 2018-01-29:
I still didn‘t solve this problem,  so I switched it to my ubuntu computer.

HaRo87, 2018-01-31:
Hey @YanMinge,
I've switched to Ubuntu 16.04 and did a fresh install following all the steps in the tutorial and I was able to call `wrap.py` without the error mentioned above. But still no luck, now I am getting a _WrapException_.

lisaong, 2018-02-15:
On macOS Sierra, I managed to get around the `PyThreadState_Get: no current thread` error by using Homebrew's version of python3 (and libs) instead of Anaconda's.

1. brew install python3
2. Run cmake and the spew should look like:
```
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.6.4", minimum required is "3.4") 
-- Found PythonLibs: /usr/local/Frameworks/Python.framework/Versions/3.6/lib/libpython3.6m.dylib (found suitable version "3.6.4", minimum required is "3.4") 
```
3. Build ELL as usual.
4. When running tools like wrap.py, call `python3` instead of `python` (the latter defaults to python2.7). 

This workaround doesn't use virtualenvs, which means stuff like numpy will need to be installed via `pip3 install numpy`, instead of `conda install numpy`. 

Somehow, the Anaconda libs and Homebrew's libs are incompatible.

lovettchris, 2018-07-18:
closing due to inactivity

closed 2018-07-18 by lovettchris