microsoft/ELL #136: Inference speed on Rpi

braca51e, 2018-03-23:
Hi,

I have a doubt about the speeds you are reporting in your tutorials. Comparing image classification models side by side on the Raspberry Pi result images claim 26 and 21 ms for mean inference speed. When I run and follow the instruction there I see much more than that 800 to 900 ms?

byronChanguion, 2018-03-26:
You are right to be suspicious: The screenshot at the top of the "Comparing image classification models side by side on the Raspberry Pi" seems to be from a PC and not on a Raspberry Pi. I will update the screenshot to one from the Raspberry Pi, sorry for the confusion.

For the time per frame on a Pi, please use the time listed for the model in the Gallery.

For old models that are no longer listed in the Gallery's table of recommended models, you can get the results by looking at the validation_p3.out from the model's directpory in the ELL-Models repo. For example, the models in the Comparing image classification models tutorial can be found at:
https://github.com/Microsoft/ELL-models/blob/master/models/ILSVRC2012/d_I160x160x3CMCMCMCMCMCMC1AS/validation_pi3.out
and
https://github.com/Microsoft/ELL-models/blob/master/models/ILSVRC2012/d_I160x160x3CMCMBMBMBMBMC1AS/validation_pi3.out

However, 800 to 900ms per frame does seem to be excessively slow. Do you have OpenBLAS enabled in your CMake environment?

braca51e, 2018-03-27:
Thanks @byronChanguion. I'm using blas, at least thats it's found by cmake. But inference times are not consistent, sometimes I get 400 ms then I restart the demo and get up to 1.5 secs not yet found a reason for this large difference. I'm using same config as listed in you Rpi set-up.

braca51e, 2018-03-27:
@byronChanguion Finally found the problem. I was using WiFi as soon as I shut it down inference speed was exactly the same that you report!!

kernhanda, 2018-04-13:
Thanks for letting us know what the problem was, @braca51e. Closing this issue now.

closed 2018-04-13 by kernhanda