microsoft/ELL #139: Slicer Build errors

SaimaSafdar, 2018-04-19:
Hello there,
I am using Visual studio 2017, git2.16.2-64-bit, cmake 3.11.0, qt 4.7.4, slicer 4.9.0, sil.1.9.7 and window 7

Please see the below:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	MSB6006	"cmd.exe" exited with code 1.	curl	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncher.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncherW.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncher.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncherW.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	MSB6006	"cmd.exe" exited with code 1.	python	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-appdirs	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-pyparsing	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-six	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-packaging	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	python-setuptools	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-smmap	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	python-gitdb	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-wheel	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-GitPython	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	python-nose	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-couchdb	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-chardet	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-pip	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	NUMPY	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	VTKv9	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3. [C:\SLicer4Build\ITKv4-build\Wrapping\Modules\ITKCommon\ITKCommonSwig.vcxproj]	ITKv4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	C1083	Cannot open include file: 'patchlevel.h': No such file or directory [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyBase\ITKPyBasePython.vcxproj]	ITKv4	c:\slicer4build\itkv4\wrapping\generators\python\patchedpython27pyconfig.h	2	
Error	C1083	Cannot open include file: 'patchlevel.h': No such file or directory [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyBase\ITKPyBasePython.vcxproj]	ITKv4	c:\slicer4build\itkv4\wrapping\generators\python\patchedpython27pyconfig.h	2	
Error	MSB6006	"cmd.exe" exited with code 1. [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyUtils\ITKPyUtilsCastXML.vcxproj]	ITKv4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3. [C:\SLicer4Build\ITKv4-build\Wrapping\Modules\ITKCommon\ITKCommonSwig.vcxproj]	ITKv4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	C1083	Cannot open include file: 'patchlevel.h': No such file or directory [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyBase\ITKPyBasePython.vcxproj]	ITKv4	c:\slicer4build\itkv4\wrapping\generators\python\patchedpython27pyconfig.h	2	
Error	C1083	Cannot open include file: 'patchlevel.h': No such file or directory [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyBase\ITKPyBasePython.vcxproj]	ITKv4	c:\slicer4build\itkv4\wrapping\generators\python\patchedpython27pyconfig.h	2	
Error	MSB6006	"cmd.exe" exited with code 1. [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyUtils\ITKPyUtilsCastXML.vcxproj]	ITKv4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
[... cut ...] nqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\s
...

kernhanda, 2018-04-20:
Can you explain a little bit about what you're trying to do? From the logs provided, ELL doesn't seem to be mentioned at all.

kernhanda, 2018-04-22:
How does slicer relate to ELL?

lovettchris, 2018-04-23:
Surely this is a misdirected post.  Please find the correct Slicer4 github, thanks.

closed 2018-04-23 by lovettchris