microsoft/ELL #141: retargetTrainer.exe crashing

CalderaSrv, 2018-04-21:
Hello,

retargetTrainer.exe is crashing in the new 2.3.2 release. 

My system:
Windows10, i7

Example command:
S:\ELL-2.3.2\build\bin\release\retargetTrainer --maxEpochs 1000 --multiClass true --refineIterations 1 --inputModelFilename pretrained.ell --targetPortElements 1448.output --inputDataFilename MyTrain.gsdf --outputModelFilename model.ell

Output:
Redirected output for port elements 1448.output from model
Then it crashes:
[image]

I have tested on these models and observe the same behavior:
dsf_I64x64x3CCMCCMCCMCMCMC1AS.ell, --targetPortElements 1448.output
Candlenut.ell --targetPortElements 1891.output
PandanFlower.ell --targetPortElements 3295.output

I am able to run without errors if I swap in my retargetTrainer.exe from 2.3.0 into my 2.3.2 \ELL-2.3.2\build\bin\Release\ folder

lovettchris, 2018-06-12:
I've confirmed this crash, we are investigating...

lovettchris, 2018-06-12:
Found the problem, the DataLoader.tcc uses old prototype for callback functions.  The callbacks now take a void* context parameter, the fix will be included in next update.

kernhanda, 2018-07-17:
Resolved with release [v2.3.5](https://github.com/Microsoft/ELL/tree/v2.3.5).

closed 2018-07-17 by kernhanda