microsoft/ELL #145: How to build on Raspbian Stretch for python 3.5?

BretStateham, 2018-05-07:
We are using ELL on Raspbian stretch, but we have a requirement for python 3.5 instead of the 3.4 installed by Miniconda3. 

Is there a way for us to get ELL on Raspbian stretch without using Miniconda3?  Can I just build it? If so, how?

Bret

lovettchris, 2018-05-07:
This is how I built it for miniconda's python 3.4.

If you modify the paths to python below to your python 3.5 install then you should get a version that works.  CMake takes a while because it is looking for all sorts of stuff, when it is done you should see a Python3 section in the output that is populated with all the right information.  If the Python3 section is blank, then something went wrong, and you need to fix your python paths until cmake finds it.  Each time cmake fails wipe your build folder so that it starts from scratch.  

https://docs.opencv.org/master/d7/d9f/tutorial_linux_install.html 
https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/
sudo apt-get install -y build-essential cmake pkg-config
sudo apt-get install -y libjpeg-dev libtiff5-dev libjasper-dev libpng12-dev libdc1394-22-dev
sudo apt-get install -y libavcodec-dev libavformat-dev libswscale-dev libv4l-dev
sudo apt-get install -y libxvidcore-dev libx264-dev
sudo apt-get install -y libgtk2.0-dev libgtk-3-dev
sudo apt-get install -y libatlas-base-dev gfortran
sudo apt-get install -y python3-dev
conda install jinja2

cd ~
wget -O opencv.zip https://github.com/Itseez/opencv/archive/3.3.0.zip
unzip opencv.zip
cd opencv-3.3.0
wget -O opencv_contrib.zip https://github.com/Itseez/opencv_contrib/archive/3.3.0.zip
unzip opencv_contrib.zip

mkdir build
cd build

cmake -D CMAKE_BUILD_TYPE=RELEASE \
-D CMAKE_INSTALL_PREFIX=/usr/local \
-D OPENCV_EXTRA_MODULES_PATH=../opencv_contrib-3.3.0/modules \
-D BUILD_EXAMPLES=OFF \
-D BUILD_TESTS=OFF -D BUILD_PERF_TESTS=OFF \
-D PYTHON3_EXECUTABLE=/home/pi/miniconda3/envs/py34/bin/python3.4 \
-D PYTHON3_INCLUDE_DIR=/home/pi/miniconda3/envs/py34/include/python3.4m \
-D PYTHON3_LIBRARY=/home/pi/miniconda3/envs/py34/lib/libpython3.4m.so \
-D PYTHON3_NUMPY_INCLUDE_DIRS=/home/pi/miniconda3/envs/py34/lib/python3.4/site-packages/numpy/core/include \
-D PYTHON3_PACKAGES_PATH=/home/pi/miniconda3/envs/py34/lib/python3.4/site-packages \
..

make

sudo make install

python3 -c "import cv2; print(cv2.__version__)"

BretStateham, 2018-05-08:
Thanks Chris, I used what you posted above, tweaked it for the Python 3.5 install that shipped with the 2018-04-18 Raspbian Jessie image, and was able to make it work.  I think we can close this issue.

Here's exactly what I did:

# Building Open CV 3.3.0 on Raspbian Stretch with Python 3.5.3

1. Starting from a clean [Raspbian Stretch image - 2018-04-18](http://downloads.raspberrypi.org/raspbian/images/raspbian-2018-04-19/2018-04-18-raspbian-stretch.zip)

1.  Imaged the SD card using the image above using [Etcher.io](http://etcher.io)

1.  Logged in as the user `pi` with the password `raspberry`

1.  Update the os:

    ```bash
    sudo apt-get update
    sudo apt-get upgrade -y
    ```
1.  Configured wifi by editing `/etc/wpa_supplicant/wpa_supplicant.conf`.  See [Headless Pi - Configure the WiFi Network](https://github.com/BretStateham/headlesspi#configure-the-wifi-network) for more info.

1. Verified the Python 3 version by running:

      ```bash
      python3 --version
      ```

      And verifying it is Python 3.5.3

      ```bash
      Python 3.5.3
      ```

1.  Install pre-reqs:

    ```bash
    sudo apt-get install -y build-essential cmake pkg-config
    sudo apt-get install -y libjpeg-dev libtiff5-dev libjasper-dev libpng12-dev libdc1394-22-dev
    sudo apt-get install -y libavcodec-dev libavformat-dev libswscale-dev libv4l-dev
    sudo apt-get install -y libxvidcore-dev libx264-dev
    sudo apt-get install -y libgtk2.0-dev libgtk-3-dev
    sudo apt-get install -y libatlas-base-dev gfortran
    sudo apt-get install -y python3-dev
    ```
1.  Upgrade pip3

    ```bash
    python3 -m pip install -U pip
    ```

1.  The update works, but somehow breaks the pathing for pip3.  To verify the new version you need to call the pip module from within python3.  Run the following command:

    ```bash
    python3 -m pip --version
    ```

    And verify that it is version 10.0.1 or greater

    ```bash
    pip 10.0.1 from /home/pi/.local/lib/python3.5/site-packages/pip (python 3.5)
    ```

1.  Install Jinja2 (remember all pip3 installs have to be run as `python3 -m pip install ...`)

    ```bash
    python3 -m pip install jinja2
    ```

1.  Install numpy

    ```bash
[... 187 lines cut ...]
    --
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/pi/opencv-3.3.0/build
    ```

1.  Run `make`

    > **Note**: This took about one hour and fourty five minutes (1:45) on my Raspberry Pi with a [32GB San Disk Ultra microSDHC UHS-I Card](http://a.co/bg5JjKF)

    ```bash
    make
    ```

1.  Run `make install`

    ```bash
    sudo make install
    ```

1.  Test it:

    ```bash
    python3 -c "import cv2; print(cv2.__version__)"
    ```

closed 2018-05-08 by BretStateham

lovettchris, 2018-05-10:
Thanks for the write up, I'll promote this to our docs and give you credit.  I'm curious about the pip3 upgrade, what problems did you run into without this upgrade?

BretStateham, 2018-05-11:
Regarding the pip3 upgrade, once you do it, if you try to run `pip3` by itself, you get the following error:

```bash
Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    from pip import main
ImportError: cannot import name 'main'
```

However, if you run pip3 (pip) by loading it as a module into python3, it works. 

For example, this command will error after you upgraded pip3:

```bash
pip3 --version
```

However, this will work:

```bash
python3 -m pip --version
```