microsoft/ELL #147: Unknown processor, disabling OpenBLAS -- BLAS library not found

Einzelkampfer, 2018-05-09:
**I am building ELL on Windows 10, with Visual Studio 2015. When I run "cmake -G "Visual Studio 14 2015 Win64" ..", it get a message about OpenBLAS:

    -- Unknown processor, disabling OpenBLAS
    -- BLAS library not found"

Then I run "cmake --build . --config Release" to build the project, and it get 60 linking errors, like:

	"C:\Users\t-suzhen\Desktop\ELL\build\ALL_BUILD.vcxproj" (default target) (1) ->
	"C:\Users\t-suzhen\Desktop\ELL\build\tools\utilities\apply\apply.vcxproj" (default target) (3) ->
	(Link target) ->
	  dsp.lib(WinogradConvolution.obj) : error LNK2019: unresolved external symbol "void __cdecl ell::math::Blas::Gemm(enum
	 ell::math::MatrixLayout,enum ell::math::MatrixTranspose,enum ell::math::MatrixTranspose,int,int,int,float,float const
	*,int,float const *,int,float,float *,int)" (?Gemm@Blas@math@ell@@YAXW4MatrixLayout@23@W4MatrixTranspose@23@1HHHMPEBMH2
	HMPEAMH@Z) referenced in function "void __cdecl ell::dsp::ComputeTransformedOutput<float>(class ell::math::ConstTensorR
	eference<float,2,1,0> const &,class ell::math::ConstTensorReference<float,2,1,0> const &,int,int,int,int,int,int,class
	ell::math::Tensor<float,2,1,0> &)" (??$ComputeTransformedOutput@M@dsp@ell@@YAXAEBV?$ConstTensorReference@M$01$00$0A@@ma
	th@1@0HHHHHHAEAV?$Tensor@M$01$00$0A@@31@@Z) [C:\Users\t-suzhen\Desktop\ELL\build\tools\utilities\apply\apply.vcxproj]
	  dsp.lib(WinogradConvolution.obj) : error LNK2019: unresolved external symbol "void __cdecl ell::math::Blas::Gemm(enum
	 ell::math::MatrixLayout,enum ell::math::MatrixTranspose,enum ell::math::MatrixTranspose,int,int,int,double,double cons
	t *,int,double const *,int,double,double *,int)" (?Gemm@Blas@math@ell@@YAXW4MatrixLayout@23@W4MatrixTranspose@23@1HHHNP
	EBNH2HNPEANH@Z) referenced in function "void __cdecl ell::dsp::ComputeTransformedOutput<double>(class ell::math::ConstT
	ensorReference<double,2,1,0> const &,class ell::math::ConstTensorReference<double,2,1,0> const &,int,int,int,int,int,in
	t,class ell::math::Tensor<double,2,1,0> &)" (??$ComputeTransformedOutput@N@dsp@ell@@YAXAEBV?$ConstTensorReference@N$01$
	00$0A@@math@1@0HHHHHHAEAV?$Tensor@N$01$00$0A@@31@@Z) [C:\Users\t-suzhen\Desktop\ELL\build\tools\utilities\apply\apply.v
	cxproj]
	  C:\Users\t-suzhen\Desktop\ELL\build\bin\Release\apply.exe : fatal error LNK1120: 2 unresolved externals [C:\Users\t-s
	uzhen\Desktop\ELL\build\tools\utilities\apply\apply.vcxproj]

I think this has something to do with the previous message, but I can't find out where the problem is. Is this because of the CPU of my computer is not supported?

lovettchris, 2018-05-09:
This is probably caused by our [CMake\OpenBLASSetup.cmake](https://github.com/Microsoft/ELL/blob/master/CMake/OpenBLASSetup.cmake) script is not recognizing your CPU id (see lines 100-160).  We only packaged two of the most common Intel CPU types in our OpenBLAS Nuget package (..\external\OpenBLASLibs.0.2.19.3\build\native\x64), Haswell and Sandybridge.  Now turns out these also work for a bunch of newer chips, which is what OpenBLASSetup.cmake is doing (for example it maps Broadwell to Haswell).  But when newer and newer chips come out we need to keep updating that mapping... So you can see that whole cmake script is operating on the output of the following registry values:
```
reg query HKEY_LOCAL_MACHINE\Hardware\Description\System\CentralProcessor\0
```
So if you attach the output of that to this issue we can work from there...

Einzelkampfer, 2018-05-10:
Thank you. My output result of this command is:

    HKEY_LOCAL_MACHINE\Hardware\Description\System\CentralProcessor\0
    Component Information    REG_BINARY    00000000000000000000000000000000
    Identifier    REG_SZ    Intel64 Family 6 Model 61 Stepping 4
    Configuration Data    REG_FULL_RESOURCE_DESCRIPTOR    FFFFFFFFFFFFFFFF0000000000000000
    ProcessorNameString    REG_SZ    Intel(R) Core(TM) i7-5600U CPU @ 2.60GHz
    VendorIdentifier    REG_SZ    GenuineIntel
    FeatureSet    REG_DWORD    0x3d1b3fff
    ~MHz    REG_DWORD    0xa22
    Update Revision    REG_BINARY    000000001F000000
    Update Status    REG_DWORD    0x0
    Previous Update Revision    REG_BINARY    000000001D000000
    Platform Specific Field 1    REG_DWORD    0x40

Einzelkampfer, 2018-05-10:
I solved the problem by adding this statements to OpenBLASSetup.cmake:

    elseif(processor_model EQUAL 61)
        set(processor_generation "haswell") 

And the build process is completed.

Einzelkampfer, 2018-05-10:
But unfortunately, not all the tests have passed when I try to build RUN_TESTS:

     The following tests FAILED:
         15 - cntk_importer_test (Failed)
         35 - wrap-test (Failed)
      Errors while running CTest

Einzelkampfer, 2018-05-11:
When I run the "python <ELL-root>/tools/wrap/wrap.py featurizer_16k.ell --target host --outdir featurizer_16k --module_name mfcc" command in the "Audio Classification Tutorials in Python", it gets warp errors:
    
    compiling model...
        command C:/Users/t-suzhen/Desktop/ELL/build/bin/release/compile failed with error code 1
        ### WrapException: <class 'buildtools.EllBuildToolsRunException'>:C:/Users/tsuzhen/Desktop/ELL/build/bin/release/compile -imap featurizer_16k.ell -cfn Predict -cmn featurizer_16k --bitcode --target host -od host --fuseLinearOps True --swig --blas true --optimize true

I think this has something to do with the warp error in RUN_TESTS.

salah-selim, 2018-05-13:
the file "featureizer_16k.ell" is missing the following:
"inputLayout1"
"inputLayout2"
"outputLayout"
For the node:
      "_type": "BinaryOperationNode<float>",
near the end of the file.
The modified file (which works without errors) is attached.
[featurizer_16k .txt](https://github.com/Microsoft/ELL/files/1998763/featurizer_16k.txt)

Einzelkampfer, 2018-05-14:
It worked. Thank you!

closed 2018-05-14 by Einzelkampfer