microsoft/ELL #148: load the Python module representing the wrapped ELL model.

vipul0110, 2018-05-11:
After competing all instruction we got following error: 

(py36) C:\Users\Vipul\Embeded\ELL>python call_model.py
Traceback (most recent call last):
  File "host\model.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Users\Vipul\AppData\Local\conda\conda\envs\py36\lib\importlib\__init_
_.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: DLL load failed: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "call_model.py", line 19, in <module>
    import model
  File "host\model.py", line 17, in <module>
    _model = swig_import_helper()
  File "host\model.py", line 16, in swig_import_helper
    return importlib.import_module('_model')
  File "C:\Users\Vipul\AppData\Local\conda\conda\envs\py36\lib\importlib\__init_
_.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: The specified module could not be found.

lovettchris, 2018-06-12:
I assume you are following all the instructions in [this tutorial](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/) ?

lovettchris, 2018-07-17:
This also happens if you did add the openblass dll location to your PATH.  See **Path Environment** in [Windows Setup](https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md).

closed 2018-07-17 by lovettchris

ankit3600, 2019-07-28:
I am also getting the same error on windows machine. I tried with/without (both) the OpenBLASLibs bin path
set PATH=%PATH%;C:\ELL\ELL\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\bin

Traceback (most recent call last):
  File "run_classifier.py", line 118, in <module>
    args.sample_rate, args.speaker, args.auto_scale, args.reset)
  File "run_classifier.py", line 59, in test_keyword_spotter
    predictor = classifier.AudioClassifier(classifier_model, categories, threshold, SMOOTHING)
  File "C:\ELL\ELL\model\classifier.py", line 60, in __init__
    self.model = ell.CompiledModel(model_path)
  File "C:\ELL\ELL\model\compiled_ell_model.py", line 29, in __init__
    self.module = importlib.import_module(model_name)
  File "C:\Users\anmish\AppData\Local\Continuum\miniconda3\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "compiled_classifier\model.py", line 11, in <module>
    import _model
ImportError: DLL load failed: The specified module could not be found.

I am following the below tutorial:
https://microsoft.github.io/ELL/tutorials/Getting-started-with-audio-keyword-spotting-on-the-Raspberry-Pi/