microsoft/ELL #149: VS 2017 build error.

Einzelkampfer, 2018-05-12:
I build the project in Windows 10 with VS 2017. When I run the "" command, most of the sub-projects are successfully built, but "dataFiles" is failed with compile error. It shows error in the file "iterator_test.cpp":

	ell\libraries\utilities\test\src\iterator_test.cpp(34): error C2668: 'ell::utilities::MakeStlContainerReferenceIterator': ambiguous call to overloaded function
	e:\ell\libraries\utilities\test\src\iterator_test.cpp(38): error C3536: 'it': cannot be used before it is initialized
	2>e:\ell\libraries\utilities\test\src\iterator_test.cpp(38): error C2228: left of '.IsValid' must have class/struct/union
	2>e:\ell\libraries\utilities\test\src\iterator_test.cpp(38): note: type is 'int'
	2>e:\ell\libraries\utilities\test\src\iterator_test.cpp(40): error C2228: left of '.Get' must have class/struct/union
	2>e:\ell\libraries\utilities\test\src\iterator_test.cpp(40): note: type is 'int'
	2>e:\ell\libraries\utilities\test\src\iterator_test.cpp(41): error C2228: left of '.Next' must have class/struct/union
	2>e:\ell\libraries\utilities\test\src\iterator_test.cpp(41): note: type is 'int'

gegogabriel, 2018-05-14:
up. I have the same problem. One notable change that i made reading through the other issues was to change in OPENBLASsetup cmake to accept my processor

Reading the comments in the cmake i saw that sky lake would be compatible with sandy so I tried with it(I also tried with the other) but the same error so I don't think this is related.

  Component Information    REG_BINARY    00000000000000000000000000000000
    Identifier    REG_SZ    Intel64 Family 6 Model 94 Stepping 3
    Configuration Data    REG_FULL_RESOURCE_DESCRIPTOR    FFFFFFFFFFFFFFFF0000000000000000
    ProcessorNameString    REG_SZ    Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
    VendorIdentifier    REG_SZ    GenuineIntel
    FeatureSet    REG_DWORD    0x3d1b3fff
    ~MHz    REG_DWORD    0xa20
    Update Revision    REG_BINARY    00000000BA000000
    Update Status    REG_DWORD    0x0
    Previous Update Revision    REG_BINARY    0000000074000000
    Platform Specific Field 1    REG_DWORD    0x20

clovett, 2018-07-10:
what happens if you run the "rebuild.cmd" script instead?

kernhanda, 2018-07-17:
Resolved with release [v2.3.5](https://github.com/Microsoft/ELL/tree/v2.3.5)

closed 2018-07-17 by kernhanda