microsoft/ELL #150: Build failed when building RUN_TESTS

Einzelkampfer, 2018-05-14:
I successfully built the ALL_BUILD and ELL_PYTHON project. When I try to build RUN_TEST, it gets following error: 

     The following tests FAILED:
             15 - cntk_importer_test (Failed)
             35 - wrap-test (Failed)
      Errors while running CTest
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: The command "
    setlocal\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: "C:\Program F
    iles\CMake\bin\ctest.exe" --force-new-ctest-process -C Release\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj
    ]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
    l% neq 0 goto :cmEnd\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmEnd\r [C:\
    Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: endlocal & ca
    ll :cmErrorLevel %errorlevel% & goto :cmDone\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmErrorLevel
    \r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: exit /b %1\r
    [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmDone\r [C:
    \Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
    l% neq 0 goto :VCEnd\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :VCEnd" exite
    d with code 8. [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    Done Building Project "C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj" (default targets) -- FAILED.

    Build FAILED.

    "C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj" (default target) (1) ->
    (PostBuildEvent target) ->
      C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: The command
     "setlocal\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: "C:\Program F
    iles\CMake\bin\ctest.exe" --force-new-ctest-process -C Release\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj
    ]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
    l% neq 0 goto :cmEnd\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmEnd\r [C:\
    Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: endlocal & ca
    ll :cmErrorLevel %errorlevel% & goto :cmDone\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmErrorLevel
    \r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: exit /b %1\r
    [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmDone\r [C:
    \Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorleve
    l% neq 0 goto :VCEnd\r [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]
    C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :VCEnd" exite
    d with code 8. [C:\Users\t-suzhen\Desktop\ELL\build\RUN_TESTS.vcxproj]

        0 Warning(s)
        1 Error(s)

clovett, 2018-07-10:
Those are the tests that depend on Python.  Try launching the ell.sln from your Anaconda Command Prompt so that the ctest process can find the right Python environment.  This is what I see when I do that:

```
1>------ Build started: Project: RUN_TESTS, Configuration: Release x64 ------
1>Test project D:/git/ELL/ELL/build
1>      Start  1: common_test
1> 1/39 Test  #1: common_test ......................   Passed    0.20 sec
1>      Start  2: data_test
1> 2/39 Test  #2: data_test ........................   Passed    0.02 sec
1>      Start  3: dsp_test
1> 3/39 Test  #3: dsp_test .........................   Passed    5.09 sec
1>      Start  4: emitters_test
1> 4/39 Test  #4: emitters_test ....................   Passed    0.17 sec
1>      Start  5: evaluators_test
1> 5/39 Test  #5: evaluators_test ..................   Passed    0.01 sec
1>      Start  6: functions_test
1> 6/39 Test  #6: functions_test ...................   Passed    0.02 sec
1>      Start  7: math_test
1> 7/39 Test  #7: math_test ........................   Passed    0.05 sec
1>      Start  8: model_test
1> 8/39 Test  #8: model_test .......................   Passed    0.03 sec
1>      Start  9: model_compiler_test
1> 9/39 Test  #9: model_compiler_test ..............   Passed    4.70 sec
1>      Start 10: nodes_test
1>10/39 Test #10: nodes_test .......................   Passed    0.05 sec
1>      Start 11: dsp_nodes_test
1>11/39 Test #11: dsp_nodes_test ...................   Passed   20.34 sec
1>      Start 12: nn_nodes_test
1>12/39 Test #12: nn_nodes_test ....................   Passed    0.03 sec
1>      Start 13: passes_test
1>13/39 Test #13: passes_test ......................   Passed    0.03 sec
1>      Start 14: predictors_test
1>14/39 Test #14: predictors_test ..................   Passed    0.02 sec
1>      Start 15: trainers_test
1>15/39 Test #15: trainers_test ....................   Passed    0.01 sec
1>      Start 16: utilities_test
1>16/39 Test #16: utilities_test ...................   Passed    0.90 sec
1>      Start 17: cntk_importer_test
1>17/39 Test #17: cntk_importer_test ...............   Passed   97.27 sec
1>      Start 18: darknet_importer_test
1>18/39 Test #18: darknet_importer_test ............   Passed    1.43 sec
1>      Start 19: forestTrainer_test
1>19/39 Test #19: forestTrainer_test ...............   Passed    0.02 sec
1>      Start 20: linearTrainer_test_0
1>20/39 Test #20: linearTrainer_test_0 .............   Passed    0.02 sec
1>      Start 21: linearTrainer_test_1
1>21/39 Test #21: linearTrainer_test_1 .............   Passed    0.03 sec
1>      Start 22: linearTrainer_test_2
1>22/39 Test #22: linearTrainer_test_2 .............   Passed    0.02 sec
1>      Start 23: linearTrainer_test_3
1>23/39 Test #23: linearTrainer_test_3 .............   Passed    0.02 sec
1>      Start 24: linearTrainer_test_4
1>24/39 Test #24: linearTrainer_test_4 .............   Passed    0.02 sec
1>      Start 25: linearTrainer_test_5
1>25/39 Test #25: linearTrainer_test_5 .............   Passed    0.02 sec
1>      Start 26: linearTrainer_test_6
1>26/39 Test #26: linearTrainer_test_6 .............   Passed    0.02 sec
1>      Start 27: linearTrainer_test_7
1>27/39 Test #27: linearTrainer_test_7 .............   Passed    0.02 sec
1>      Start 28: linearTrainer_test_8
1>28/39 Test #28: linearTrainer_test_8 .............   Passed    0.02 sec
1>      Start 29: linearTrainer_test_9
1>29/39 Test #29: linearTrainer_test_9 .............   Passed    0.02 sec
1>      Start 30: linearTrainer_test_10
1>30/39 Test #30: linearTrainer_test_10 ............   Passed    0.02 sec
1>      Start 31: protoNNTrainer_test_0
1>31/39 Test #31: protoNNTrainer_test_0 ............   Passed    0.14 sec
1>      Start 32: sweepingSGDTrainer_test
1>32/39 Test #32: sweepingSGDTrainer_test ..........   Passed    0.02 sec
1>      Start 33: apply_test
1>33/39 Test #33: apply_test .......................   Passed    0.02 sec
1>      Start 34: compile_test1
1>34/39 Test #34: compile_test1 ....................   Passed    0.03 sec
1>      Start 35: compile_test2
1>35/39 Test #35: compile_test2 ....................   Passed    0.03 sec
1>      Start 36: make_profiler_test
1>36/39 Test #36: make_profiler_test ...............   Passed   33.71 sec
1>      Start 37: wrap-test
1>37/39 Test #37: wrap-test ........................   Passed   37.28 sec
1>      Start 38: ell-python-interface-test
1>38/39 Test #38: ell-python-interface-test ........   Passed    1.67 sec
1>      Start 39: ell_module_test
1>39/39 Test #39: ell_module_test ..................   Passed    0.16 sec
1>
1>100% tests passed, 0 tests failed out of 39
1>
1>Total Test time (real) = 203.94 sec
========== Build: 1 succeeded, 0 failed, 1 up-to-date, 0 skipped ==========

```

lovettchris, 2018-07-18:
We just pushed new bits, can you try again with the latest? Thanks.

closed 2018-07-18 by lovettchris