microsoft/ELL #156: c++ api multiple definition of `model_InputCallback'

gegogabriel, 2018-05-19:
Hi the example from the tutorial works great. 
However when I'm wrapping the code in a class(.h + cpp) I get this error + multiple definition of model_InputCallback. Where the include "model.h" is defined in the header of the class.

I have no idea how I can get this working. Thanks in advance.
-------------
Relevant log

[build] CMakeFiles/tutorial.dir/tutorial_jpeg.cpp.o: In function `cvflann::anyimpl::small_any_policy<unsigned int>::static_delete(void**)':
[build] /home/gabriel/Documents/tut_desk_build/model/model.h:115: multiple definition of `model_InputCallback'
[build] CMakeFiles/tutorial.dir/wrap.cpp.o:/home/gabriel/Documents/tut_desk_build/model/model.h:115: first defined here
[build] CMakeFiles/tutorial.dir/tutorial_jpeg.cpp.o: In function `model_OutputCallback':
[build] /home/gabriel/Documents/tut_desk_build/model/model.h:121: multiple definition of `model_OutputCallback'
[build] CMakeFiles/tutorial.dir/wrap.cpp.o:/home/gabriel/Documents/tut_desk_build/model/model.h:121: first defined here
[build] collect2: error: ld returned 1 exit status

gegogabriel, 2018-05-20:
I figured it would be hard to understand from the intial description, so:
I made a gist with the relavant files.

In the past day I tried everything I know and could find on the internet, however whatever I'm doing it just doesen't work.
I want to mention that what is very unusual is that if I have only .h files including model.h it's ok. I made , a new class in a /h wrapping functions from model.h and I could use it in main.cpp. 
Howver in the moment when I'm including the wrapper.h in another .h which has also .cpp with implementations it gives me these error.

Any help will me much appreciated.

cjacobs, 2018-05-30:
Hi,

This error is due to the implementations of the `model_InputCallback` and `model_OutputCallback` being in the model header file. If the header gets included more than once, those callback functions can get multiply defined.

There's a fix in the works for this error, and it should appear here soon. In the meantime, here's a workaround:

At the top of the generated `model.h` file, add a `#pragma once` to ensure the file doesn't get included multiple times. Then at the bottom of the same file, put an `#ifdef` block around the callback definitions:

```
#pragma once

...

#ifdef MAIN
extern "C" 
{
  // (callbacks here)
} // extern "C"
#endif
```

Then in your main.cpp file, define the `MAIN` symbol before including the model header file:

```
#define MAIN
#include "model.h"

...
```

Now the callback definitions will only be seen when compiling `main.cpp`, and the linker error should go away.

Keep in mind that the `model.h` file gets regenerated when you compile the model, so you'll have to re-modify it if you recompile.

Hope this helps!

--chuck

closed 2018-05-30 by cjacobs