microsoft/ELL #158: Package opencv from your channel in anaconda is missing

toni-pad, 2018-05-25:
Hello,

When I try to set up my RPi3 with the line
$ conda install -c microsoft-ell opencv -y
The result is 
Error: No packages found in current linux-armv7l channels matching: opencv

Yes, I can see the package with anaconda search -t conda opencv
     microsoft-ell/opencv      |          | conda           | linux-armv6l, linux-armv7l
And anaconda show microsoft-ell/opencv shows
Using binstar api site https://api.anaconda.org
Name:    opencv
Summary: 
Access:  public
Package Types:  conda
Versions:
   + 3.2.0
   + 3.3.0

But the error persists. Can I install other opencv for example from poppy-project?

cornellnikko, 2018-06-05:
Hi toni-pad,

I looked around the Anaconda website and saw that on https://anaconda.org/microsoft-ell/opencv the following commands are suggested:

``conda install -c microsoft-ell/label/stretch opencv``

Although this installs fine, I get dragged back into dependency hell when I try to run tutorial.py:

``File "tutorial.py", line 13, in <module>``
``import cv2``
``ImportError: libjasper.so.1: cannot open shared object file: No such file or directory``

Hopefully that'll help, maybe someone can figure out the next steps.

clovett, 2018-06-12:
Cool, yeah, sorry about the missing libjasper, couldn't figure out how to build OpenCV with that properly included.  We removed the "default" install of OpenCV which you would get with your original command line `conda install -c microsoft-ell opencv -y` because if we provide a default then you can't install the jessie version any more, some weird conda limitation...  But you found the right path.  We have updated docs coming that will make all this more clear.

toni-pad, 2018-06-20:
Thanks cornellnikko for your hint.

I use
sudo apt install libjasper1
and 
conda install -c microsoft-ell/label/stretch opencv

Thanks.

lovettchris, 2018-07-17:
I added this to the [setup instructions](https://github.com/Microsoft/ELL/blob/master/docs/tutorials/Raspberry-Pi-setup/index.md), thanks.

closed 2018-07-17 by lovettchris