microsoft/ELL #162: Import an own CNTK model does not work

julian0001, 2018-07-06:
Since a while I am trying to import my CNTK model into the ELL environment. But the import does not work. It is only a conversion from .cntk to .ell  (https://microsoft.github.io/ELL/tutorials/Importing-models/). The ELL works for now only with the base model  VGG16_ImageNet_Caffe.model. Therefore I have tried to make my own model based on the  VGG16_ImageNet_Caffe.model (https://docs.microsoft.com/en-us/cognitive-toolkit/Build-your-own-image-classifier-using-Transfer-Learning). But I get following ouput:

$ python ELL.git/trunk/tools/importers/CNTK/cntk_import.py TransferLearning.model
-- Using new importer engine --
Loading...
Selected CPU as the process wide default device.

Finished loading.
Pre-processing...

Finished pre-processing.
Error occurred attempting to convert cntk layers to ELL model using nodes
Traceback (most recent call last):
  File "ELL.git/trunk/tools/importers/CNTK/cntk_import.py", line 106, in <module>
    main(sys.argv[1:]) # drop the first argument (program name)
  File "ELL.git/trunk/tools/importers/CNTK/cntk_import.py", line 87, in main
    ell_map = cntk_to_ell.map_from_cntk_model_using_new_engine(filename, step_interval, lag_th
  File "C:\Users\Julian\Documents\ELL\ELL.git\trunk\tools\importers\CNTK\cntk_to_ell.py", linengine
    raise exception
  File "C:\Users\Julian\Documents\ELL\ELL.git\trunk\tools\importers\CNTK\cntk_to_ell.py", linengine
    importer_model = import_nodes(cntk_nodes)
  File "C:\Users\Julian\Documents\ELL\ELL.git\trunk\tools\importers\CNTK\cntk_to_ell.py", line
    node = CntkConverterFactory.get_converter(cntk_node, cntk_converter_map)
  File "C:\Users\Julian\Documents\ELL\ELL.git\trunk\tools\importers\CNTK\lib\cntk_converters.p
    "- no converter found")
Exception: Cannot not process Pooling- no converter found
(py36)

Also when I try to use only the base model VGG16_ImageNet_Caffe.model the conversion does not work.

clovett, 2018-07-10:
This error message usually means our importer ran into a "layer" type that is not supported yet.  In this case it seems to be complaining about the "Pooling" layer, which is odd, because I'm pretty sure we do support importing CNTK Pooling layers. But if you look at the cntk_converter_map in cntk_to_ell.py you will see it lists AveragePooling and MaxPooling, but not just "Pooling" so perhaps that is the problem.  But then when I look at the [CNTK docs ](https://cntk.ai/pythondocs/layerref.html) I don't see "Pooling" listed, I only see AveragePooling and MaxPooling.  So this will take some debugging.  If you put a breakpoint on this import process can you print out more details about the CNTK layer being imported?

julian0001, 2018-07-20:
Hallo @clovett , 

I will check this now, but generally it should be possible to import a CNTK model as mentioned in the tutorial. Anyway are there any simple examples of a cntk model which I can reuse and which were already tested with the ELL. Because by now, as you mentioned, you are pretty sure that it would be supported. So I think it would be great if there are any examples with the cntk and also darknet model.

Thank you,
Julian Sprenger

lovettchris, 2018-07-20:
All these models should work https://github.com/Microsoft/ELL-models

julian0001, 2018-07-20:
Hallo @clovett ,

they do, but if I want to import my own models with CNTK, like I mentioned (https://docs.microsoft.com/en-us/cognitive-toolkit/Build-your-own-image-classifier-using-Transfer-Learning). Because the my aim is, and I think it should also be so, to train with the own dataset a model and import it in ELL.

lovettchris, 2018-07-27:
I see, yes, we still have some work to do to support every layer type that is supported by CNTK.  Thanks for the bug report, I'll make sure we take a look at the pretrained models on the [CNTK website](https://docs.microsoft.com/en-us/cognitive-toolkit/Build-your-own-image-classifier-using-Transfer-Learning).