microsoft/ELL #163: Cannot download the raw zip model files.

riple, 2018-07-16:
All the model files in .zip format cannot be downloaded. All I got from the RAW link on the .zip file page is a 133Byte description of the file property.

I even tried to download the whole repository in a zip package. (I even tried to git clone the repository to my computer). The content of the particular *.zip file is the same 133Byte description.

CalderaSrv, 2018-07-16:
I also had problems in the past week.  Specifically with the "ELL featurizer model file" and "ELL classifier model file" from the "Getting started with audio keyword classification" tutorial.  http links and curl commands download a 1kb zip file with no usable content.

riple, 2018-07-17:
> desktop:~/work/ell/ELL-models/models/ILSVRC2012/d_I224x224x3CMCMCMCMCMCMC1AS$ git lfs fetch
fetch: Fetching reference refs/heads/master
batch response: This repository is over its data quota. Purchase more data packs to restore access.                                           
batch response: This repository is over its data quota. Purchase more data packs to restore access.
error: failed to fetch some objects from 'https://github.com/Microsoft/ELL-models.git/info/lfs'

lovettchris, 2018-07-17:
Yep, sorry about that, we ran up against our bandwidth quota, we are working on a fix, apologies for the inconvenience.

lovettchris, 2018-07-21:
This is fixed now.

closed 2018-07-21 by lovettchris