microsoft/ELL #166: retargetTrainer does not create the final model.ell file

julian0001, 2018-07-22:
I am trying to run the tutorial [https://microsoft.github.io/ELL/tutorials/Repurposing-a-pretrained-image-classifier/](url). I can generate the train and validate files, but then when I try to build the model.ell file it does not do anything at all. There is no file after it and the ouput of the retargetTrainer is too less.

My ELL nodes are the following:

(py36) C:\Users\Julian\Documents\ELL\transfer_learning>..\ELL.git\trunk\build\bin\release\print.exe -imap pretrained.ell --includeNodeId --refineIterations 1
<id:1324> InputNode<double>(1)
<id:1325> ClockNode(output[0:1])
<id:1326> SourceNode<float>(output[0:2])
<id:1327> ConstantNode<float>()
<id:1328> ConstantNode<float>()
<id:1329> BroadcastLinearFunctionNode<float>(output[0:12288], output[0:3], output[0:3])
<id:1330> ConstantNode<float>()
<id:1332> BroadcastLinearFunctionNode<float>(output[0:12288], output[0:3], )
<id:1334> ConstantNode<float>()
<id:1335> BroadcastLinearFunctionNode<float>(output[0:12288], , output[0:3])
<id:1336> UnrolledConvolutionNode<float>(output[0:13068])
<id:1338> ConstantNode<float>()
<id:1339> BroadcastLinearFunctionNode<float>(output[0:131072], , output[0:32])
<id:1340> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:131072])
<id:1341> ConstantNode<float>()
<id:1342> ConstantNode<float>()
<id:1343> BroadcastLinearFunctionNode<float>(output[0:131072], output[0:32], output[0:32])
<id:1344> ConstantNode<float>()
<id:1346> BroadcastLinearFunctionNode<float>(output[0:131072], output[0:32], )
<id:1348> ConstantNode<float>()
<id:1349> BroadcastLinearFunctionNode<float>(output[0:131072], , output[0:32])
<id:1350> UnrolledConvolutionNode<float>(output[0:139392])
<id:1352> ConstantNode<float>()
<id:1353> BroadcastLinearFunctionNode<float>(output[0:131072], , output[0:32])
<id:1354> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:131072])
<id:1355> PoolingLayerNode<float,MaxPoolingFunction>(output[0:131072])
    PoolingLayer<float,MaxPoolingFunction>(shape=[64,64,32]->[32,32,32], function=maxpooling, stride=2, size=2)
<id:1356> ConstantNode<float>()
<id:1357> ConstantNode<float>()
<id:1358> BroadcastLinearFunctionNode<float>(output[0:32768], output[0:32], output[0:32])
<id:1359> ConstantNode<float>()
<id:1361> BroadcastLinearFunctionNode<float>(output[0:32768], output[0:32], )
<id:1363> ConstantNode<float>()
<id:1364> BroadcastLinearFunctionNode<float>(output[0:32768], , output[0:32])
<id:1365> UnrolledConvolutionNode<float>(output[0:36992])
<id:1367> ConstantNode<float>()
<id:1368> BroadcastLinearFunctionNode<float>(output[0:65536], , output[0:64])
<id:1369> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:65536])
<id:1370> ConstantNode<float>()
<id:1371> ConstantNode<float>()
<id:1372> BroadcastLinearFunctionNode<float>(output[0:65536], output[0:64], output[0:64])
<id:1373> ConstantNode<float>()
<id:1375> BroadcastLinearFunctionNode<float>(output[0:65536], output[0:64], )
<id:1377> ConstantNode<float>()
<id:1378> BroadcastLinearFunctionNode<float>(output[0:65536], , output[0:64])
<id:1379> UnrolledConvolutionNode<float>(output[0:73984])
<id:1381> ConstantNode<float>()
<id:1382> BroadcastLinearFunctionNode<float>(output[0:65536], , output[0:64])
<id:1383> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:65536])
<id:1384> PoolingLayerNode<float,MaxPoolingFunction>(output[0:65536])
    PoolingLayer<float,MaxPoolingFunction>(shape=[32,32,64]->[16,16,64], function=maxpooling, stride=2, size=2)
<id:1385> ConstantNode<float>()
<id:1386> ConstantNode<float>()
<id:1387> BroadcastLinearFunctionNode<float>(output[0:16384], output[0:64], output[0:64])
<id:1388> ConstantNode<float>()
<id:1390> BroadcastLinearFunctionNode<float>(output[0:16384], output[0:64], )
<id:1392> ConstantNode<float>()
<id:1393> BroadcastLinearFunctionNode<float>(output[0:16384], , output[0:64])
<id:1394> UnrolledConvolutionNode<float>(output[0:20736])
<id:1396> ConstantNode<float>()
<id:1397> BroadcastLinearFunctionNode<float>(output[0:32768], , output[0:128])
<id:1398> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:32768])
<id:1399> ConstantNode<float>()
<id:1400> ConstantNode<float>()
<id:1401> BroadcastLinearFunctionNode<float>(output[0:32768], output[0:128], output[0:128])
<id:1402> ConstantNode<float>()
<id:1404> BroadcastLinearFunctionNode<float>(output[0:32768], output[0:128], )
<id:1406> ConstantNode<float>()
<id:1407> BroadcastLinearFunctionNode<float>(output[0:32768], , output[0:128])
<id:1408> UnrolledConvolutionNode<float>(output[0:41472])
<id:1410> ConstantNode<float>()
<id:1411> BroadcastLinearFunctionNode<float>(output[0:32768], , output[0:128])
<id:1412> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:32768])
<id:1413> PoolingLayerNode<float,MaxPoolingFunction>(output[0:32768])
    PoolingLayer<float,MaxPoolingFunction>(shape=[16,16,128]->[8,8,128], function=maxpooling, stride=2, size=2)
<id:1414> ConstantNode<float>()
<id:1415> ConstantNode<float>()
<id:1416> BroadcastLinearFunctionNode<float>(output[0:8192], output[0:128], output[0:128])
<id:1417> ConstantNode<float>()
<id:1419> BroadcastLinearFunctionNode<float>(output[0:8192], output[0:128], )
<id:1421> ConstantNode<float>()
<id:1422> BroadcastLinearFunctionNode<float>(output[0:8192], , output[0:128])
<id:1423> UnrolledConvolutionNode<float>(output[0:12800])
<id:1425> ConstantNode<float>()
<id:1426> BroadcastLinearFunctionNode<float>(output[0:16384], , output[0:256])
<id:1427> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:16384])
<id:1428> PoolingLayerNode<float,MaxPoolingFunction>(output[0:16384])
    PoolingLayer<float,MaxPoolingFunction>(shape=[8,8,256]->[4,4,256], function=maxpooling, stride=2, size=2)
<id:1429> ConstantNode<float>()
<id:1430> ConstantNode<float>()
<id:1431> BroadcastLinearFunctionNode<float>(output[0:4096], output[0:256], output[0:256])
<id:1432> ConstantNode<float>()
<id:1434> BroadcastLinearFunctionNode<float>(output[0:4096], output[0:256], )
<id:1436> ConstantNode<float>()
<id:1437> BroadcastLinearFunctionNode<float>(output[0:4096], , output[0:256])
<id:1438> UnrolledConvolutionNode<float>(output[0:9216])
<id:1440> ConstantNode<float>()
<id:1441> BroadcastLinearFunctionNode<float>(output[0:8192], , output[0:512])
<id:1442> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:8192])
<id:1443> PoolingLayerNode<float,MaxPoolingFunction>(output[0:8192])
    PoolingLayer<float,MaxPoolingFunction>(shape=[4,4,512]->[2,2,512], function=maxpooling, stride=2, size=2)
<id:1444> UnrolledConvolutionNode<float>(output[0:2048])
<id:1446> ConstantNode<float>()
<id:1447> BroadcastLinearFunctionNode<float>(output[0:4000], , output[0:1000])
<id:1448> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(output[0:4000])
<id:1449> PoolingLayerNode<float,MeanPoolingFunction>(output[0:4000])
    PoolingLayer<float,MeanPoolingFunction>(shape=[2,2,1000]->[1,1,1000], function=meanpooling, stride=1, size=2)
<id:1450> SoftmaxLayerNode<float>(output[0:1000])
    SoftmaxLayer<float>(shape=[1,1,1000]->[1,1,1000])
<id:1451> ConstantNode<bool>()
<id:1452> SinkNode<float>(output[0:1000], output[0:1])
<id:1453> OutputNode<float>(output[0:1000])
<id:1445> ConstantNode<float>()
<id:1439> ConstantNode<float>()
<id:1435> ConstantNode<float>()
<id:1433> ConstantNode<float>()
<id:1424> ConstantNode<float>()
<id:1420> ConstantNode<float>()
<id:1418> ConstantNode<float>()
<id:1409> ConstantNode<float>()
<id:1405> ConstantNode<float>()
<id:1403> ConstantNode<float>()
<id:1395> ConstantNode<float>()
<id:1391> ConstantNode<float>()
<id:1389> ConstantNode<float>()
<id:1380> ConstantNode<float>()
<id:1376> ConstantNode<float>()
<id:1374> ConstantNode<float>()
<id:1366> ConstantNode<float>()
<id:1362> ConstantNode<float>()
<id:1360> ConstantNode<float>()
<id:1351> ConstantNode<float>()
<id:1347> ConstantNode<float>()
<id:1345> ConstantNode<float>()
<id:1337> ConstantNode<float>()
<id:1333> ConstantNode<float>()
<id:1331> ConstantNode<float>()

And the output of the retargetTrainer:

(py36) C:\Users\Julian\Documents\ELL\transfer_learning>..\ELL.git\trunk\build\bin\release\retargetTrainer --maxEpochs 100 --multiClass true --refineIterations 1 --inputModelFilename pretrained.ell --targetPortElements 1442.output --inputDataFilename fruit_train.gsdf --outputModelFilename model.ell
Redirected output for port elements 1442.output from model

Then there comes nothing. Is there a way to test the retargetTrainer quickly? Or could it be that I have too less training examples? Because in the example there are 400 images in use.

lovettchris, 2018-07-27:
I believe the retargetTrainer requires at least 2 classes.  If you created only one class for retraining I think the retargetTrainer gets stuck.  It shouldn't matter if you have only a small number of images in each class.

clovett, 2019-01-08:
I found a bug in the retarget trainer, a fix is on the way...

kernhanda, 2019-01-10:
Fixed with release [v2.3.7](https://github.com/Microsoft/ELL/releases/tag/v2.3.7).

closed 2019-01-10 by kernhanda