microsoft/ELL #168: ELL error

pengbo19921206, 2018-07-26:
I am Chinese.When i make it,I found the mistake.Can you help me?
../../../libraries/dsp/libdsp.a(WinogradConvolution.cpp.o): In function `void ell::dsp::ComputeTransformedOutput<float>(ell::math::ConstTensorReference<float, (ell::math::Dimension)2, (ell::math::Dimension)1, (ell::math::Dimension)0> const&, ell::math::ConstTensorReference<float, (ell::math::Dimension)2, (ell::math::Dimension)1, (ell::math::Dimension)0> const&, int, int, int, int, int, int, ell::math::Tensor<float, (ell::math::Dimension)2, (ell::math::Dimension)1, (ell::math::Dimension)0>&)':
WinogradConvolution.cpp:(.text._ZN3ell3dsp24ComputeTransformedOutputIfEEvRKNS_4math20ConstTensorReferenceIT_LNS2_9DimensionE2ELS5_1ELS5_0EEES8_iiiiiiRNS2_6TensorIS4_LS5_2ELS5_1ELS5_0EEE[_ZN3ell3dsp24ComputeTransformedOutputIfEEvRKNS_4math20ConstTensorReferenceIT_LNS2_9DimensionE2ELS5_1ELS5_0EEES8_iiiiiiRNS2_6TensorIS4_LS5_2ELS5_1ELS5_0EEE]+0x167): undefined reference to `ell::math::Blas::Gemm(ell::math::MatrixLayout, ell::math::MatrixTranspose, ell::math::MatrixTranspose, int, int, int, float, float const*, int, float const*, int, float, float*, int)'
../../../libraries/dsp/libdsp.a(WinogradConvolution.cpp.o): In function `void ell::dsp::ComputeTransformedOutput<double>(ell::math::ConstTensorReference<double, (ell::math::Dimension)2, (ell::math::Dimension)1, (ell::math::Dimension)0> const&, ell::math::ConstTensorReference<double, (ell::math::Dimension)2, (ell::math::Dimension)1, (ell::math::Dimension)0> const&, int, int, int, int, int, int, ell::math::Tensor<double, (ell::math::Dimension)2, (ell::math::Dimension)1, (ell::math::Dimension)0>&)':
WinogradConvolution.cpp:(.text._ZN3ell3dsp24ComputeTransformedOutputIdEEvRKNS_4math20ConstTensorReferenceIT_LNS2_9DimensionE2ELS5_1ELS5_0EEES8_iiiiiiRNS2_6TensorIS4_LS5_2ELS5_1ELS5_0EEE[_ZN3ell3dsp24ComputeTransformedOutputIdEEvRKNS_4math20ConstTensorReferenceIT_LNS2_9DimensionE2ELS5_1ELS5_0EEES8_iiiiiiRNS2_6TensorIS4_LS5_2ELS5_1ELS5_0EEE]+0x167): undefined reference to `ell::math::Blas::Gemm(ell::math::MatrixLayout, ell::math::MatrixTranspose, ell::math::MatrixTranspose, int, int, int, double, double const*, int, double const*, int, double, double*, int)'
collect2: error: ld returned 1 exit status
tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/build.make:161: recipe for target 'bin/makeExamples' failed
make[2]: *** [bin/makeExamples] Error 1
CMakeFiles/Makefile2:2993: recipe for target 'tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/all' failed
make[1]: *** [tools/utilities/makeExamples/CMakeFiles/makeExamples.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

lovettchris, 2018-07-27:
Undefined reference to ell::math::Blas::Gemm indicates a link error because the linker cannot find your OpenBlas library.  Looks like you are using Linux, so please double check you followed all the [Ubuntu Installation](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md) instructions, specifically this line:

```
sudo apt-get install -y libopenblas-dev
```

lovettchris, 2018-07-27:
Or if that doesn't work try and do "git clean -dfx" and "git pull" to make sure your have a clean enlistment before you build.

riple, 2018-07-29:
Look into your cmake log, it should say it cannot find the openblas library.

I am using Ubuntu 18.04 64bit, and I ran into the same issue. I fixed it by adding the installation path of OpenBlas to the CMake/OpenBLASSetup.cmake settings file.

The x86_64-linux-gnu lines are manually added.

`if(NOT WIN32)
    ## Note: libopenblas installs on ubuntu in /usr/lib and /usr/include or /opt/OpenBLAS if it's manually built
    ## Note: libopenblas installs on openSUSE in /usr/lib64 and /usr/include/openblas
    set(BLAS_INCLUDE_SEARCH_PATHS
        /opt/OpenBLAS/include
        /System/Library/Frameworks/Accelerate.framework/Versions/Current/Frameworks/vecLib.framework/Versions/Current/Headers/
        /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Accelerate.framework/Frameworks/vecLib.framework/Headers/
        /usr/local/Cellar/openblas/0.2.20_1/include
        /usr/include
        /usr/local/include
        /usr/include/openblas
        /usr/include/x86_64-linux-gnu
    )

    set(BLAS_LIB_SEARCH_PATHS
        /opt/OpenBLAS/lib
        /usr/local/Cellar/openblas/0.2.20_1/lib
        /usr/lib64/atlas-sse3
        /usr/lib64/atlas
        /usr/lib64
        /usr/local/lib64/atlas
        /usr/local/lib64
        /usr/lib/atlas-sse3
        /usr/lib/atlas-sse2
        /usr/lib/atlas-sse
        /usr/lib/atlas-3dnow
        /usr/lib/atlas
        /usr/lib
        /usr/local/lib/atlas
        /usr/local/lib
        /usr/lib/x86_64-linux-gnu/openblas
    )
endif()
`

pengbo19921206, 2018-07-30:
Thank you very much.I want to use the ELL in ARM Cortex-M4.But  ARM Cortex-M4 can not use linux，because ARM Cortex-M4  has not MMU .What should I do?

lovettchris, 2018-08-13:
You build the ELL git repo on your PC.  This produces a "compile" tool which you will find in your build/bin output folder.  This is actually a cross-compiler meaning it can compile code that runs on other platforms, as well as on your PC.  To compile your model to run on Cortex-M4 see this [Wiki page for instructions](https://github.com/Microsoft/ELL/wiki/Building-an-ELL-model-for-ARM-Cortex-M4).

closed 2018-08-28 by lovettchris