microsoft/ELL #172: Build Failed on Windows

Zhengyu-Li, 2018-08-02:
Hi all,

When I run `cmake --build . --config Release`, I got
 **`Done Building Project "D:\ELL-2.3.5\build\ALL_BUILD.vcxproj" (default targets) -- FAILED.`**
**`Build FAILED.`**

Does anyone meet the same problem?

Here is the output:

"D:\ELL-2.3.5\build\ALL_BUILD.vcxproj" (default target) (1) ->
"D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj" (default target) (39) ->
(ClCompile target) ->
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(182): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(183): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(184): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(188): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(190): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(192): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(193): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\model_compiler_test_main.cpp(194): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(1391): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(1458): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(1523): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(1601): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(1678): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(2110): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(2219): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(2253): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(2289): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(2324): error C2872: 'ConvolutionMethod': ambiguous symbol [D:\ELL-2.3.5\build\libraries\model\model_compiler_test.vcxproj]
  D:\ELL-2.3.5\libraries\model\test\src\CompilableNodesTest.cpp(2707): error C2440: 'initializing': cannot convert from 'initializer list' to 'ell::predictors::neural::RegionDetectionParameters' [D:\ELL-2.3.5\build\libraries\model\mod
el_compiler_test.vcxproj]

0 Warning(s)
19 Error(s)

Zhengyu-Li, 2018-08-03:
@kernhanda Seems everything is ok but LLVM.

(py36) D:\ELL-2.3.5\build>cmake -G "Visual Studio 14 2015 Win64" ..
-- Selecting Windows SDK version  to target Windows 10.0.17134.
-- PYTHONINTERP_FOUND=TRUE
-- PYTHON_EXECUTABLE=C:/Users/lizhengyu/AppData/Local/Continuum/anaconda3/envs/py36/python.exe
-- PYTHON_VERSION_STRING=3.6.6
-- PYTHON_VERSION_MAJOR=3
-- PYTHONLIBS_FOUND=TRUE
-- PYTHON_LIBRARIES=C:/Users/lizhengyu/AppData/Local/Continuum/anaconda3/envs/py36/libs/python36.lib
-- PYTHON_INCLUDE_PATH=C:/Users/lizhengyu/AppData/Local/Continuum/anaconda3/envs/py36/include
-- PYTHON_INCLUDE_DIRS=C:/Users/lizhengyu/AppData/Local/Continuum/anaconda3/envs/py36/include
-- PYTHON_DEBUG_LIBRARIES=PYTHON_DEBUG_LIBRARY-NOTFOUND
-- PYTHONLIBS_VERSION_STRING=3.6.6
-- Processor family: 6, model: 94
-- Using OpenBLAS compiled for haswell
-- Using BLAS include path: D:/ELL-2.3.5/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/include
-- Using BLAS library: D:/ELL-2.3.5/external/OpenBLASLibs.0.2.19.3/build/native/x64/haswell/lib/libopenblas.dll.a
-- Using BLAS DLLs: libopenblas.dll;libgcc_s_seh-1.dll;libgfortran-3.dll;libquadmath-0.dll
CMake Warning at CMake/LLVMSetup.cmake:48 (find_package):
  By not providing "FindLLVM.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LLVM", but
  CMake did not find one.

  Could not find a package configuration file provided by "LLVM" (requested
  version 3.9) with any of the following names:

    LLVMConfig.cmake
    llvm-config.cmake

  Add the installation prefix of "LLVM" to CMAKE_PREFIX_PATH or set
  "LLVM_DIR" to a directory containing one of the above files.  If "LLVM"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:118 (include)

-- LLVM not found, please check that LLVM is installed.
ERRORLLVM not found, please check that LLVM is installed.
-- Creating wrappers for python
CMake Deprecation Warning at D:/CMake/share/cmake-3.11/Modules/UseSWIG.cmake:272 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:127 (swig_add_module)
  CMake/CommonInterfaces.cmake:159 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)

-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Configuring done
-- Generating done
-- Build files have been written to: D:/ELL-2.3.5/build

kernhanda, 2018-08-03:
The warning about LLVM can be safely ignored. Please note that you're using Visual Studio 2015, per the `-G "Visual Studio 14 2015 Win64"` part of the cmake command. We no longer support VS2015. Please try it with Visual Studio 2017, which is available for free.