microsoft/ELL #173: CNTK import - activation function on fully connected layers

sdudeck, 2018-08-22:
Hello,
I am trying to import a simple cntk-model consisting of one fully connected layer only (18 inputs, 2 outputs, relu-activation).
It seems as if the ELL-import skips the activation function at all. Independend of the activation function I define for the layer (cntk.tanh, cntk.sigmoid, cntk.relu, None) the resulting ELL-json-file looks exactly the same (beside the weights).
Is this a known issue? I think my problems with importing a darknet model a couple of month ago was quite similar, that the activation layer was not imported properly.

Thank you very much,
Sven

[Model_ell.zip](https://github.com/Microsoft/ELL/files/2310196/Model_ell.zip)
[Model_cntk.zip](https://github.com/Microsoft/ELL/files/2310197/Model_cntk.zip)

lovettchris, 2018-09-03:
Yes indeed, there is a bug in CntkDenseConverter, in cntk_converters.py.   This line:
```python
if self.activation:
```
should be:
```python
if self.activation is not None:
```

I'll push a fix.

sdudeck, 2018-09-04:
thanks a lot - changing this line did the trick :-)

closed 2018-09-04 by sdudeck