microsoft/ELL #174: ModuleNotFoundError: No module named '_model' when running call_model.py

bhakthil, 2018-08-23:
I am following the Getting Started tutorial and when I am calling call_model.py, I get the following error;
**(py36) c:\Development\ELL\host>**python call_model.py
Traceback (most recent call last):
  File "c:\Development\ELL\host\model.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Anaconda3\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 948, in _find_and_load_unlocked
ModuleNotFoundError: No module named '_model'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "call_model.py", line 3, in <module>
    import model
  File "c:\Development\ELL\host\model.py", line 17, in <module>
    _model = swig_import_helper()
  File "c:\Development\ELL\host\model.py", line 16, in swig_import_helper
    return importlib.import_module('_model')
  File "C:\Anaconda3\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_model'

I am calling the py code at 'host' and have all the required python code in that directory; 
model.py
tutorial_helpers.py
__init__.py
call_model.py

I also see \build\Release\_model.pyd.
Can you please help?
Thanks

lovettchris, 2018-08-28:
"Release_model.pyd" ? I assum you are missing a backslash there and it is "\build\Release\_model.pyd" ?  Can you try this change to call_model.py:

```
from host import model
```

bhakthil, 2018-08-28:
This was due to previuos build issue. build issue was fixed by adding correct path to SWIG_LIB path.

closed 2018-08-28 by bhakthil

dougollerenshaw, 2019-06-03:
Hi, I'm also seeing the above issue after running through all of the build steps here:
https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/

Here're the contents of the 'host' directory after building:

```
(py36) OSXFJK7L-719:host dougo$ pwd
/Users/dougo/code/ell/ELL/host
(py36) OSXFJK7L-719:host dougo$ ls -l
total 207776
-rw-r--r--  1 dougo  1457291929      2268 Jun  2 22:48 CMakeLists.txt
-rw-r--r--  1 dougo  1457291929      9172 Jun  2 22:47 OpenBLASSetup.cmake
-rw-r--r--  1 dougo  1457291929       298 Jun  2 22:48 __init__.py
drwxr-xr-x  3 dougo  1457291929        96 Jun  2 22:49 __pycache__
drwxr-xr-x  7 dougo  1457291929       224 Jun  2 22:50 build
-rw-r--r--  1 dougo  1457291929       204 Jun  2 22:47 compile.log
drwxr-xr-x  2 dougo  1457291929        64 Jun  2 22:29 include
-rw-r--r--  1 dougo  1457291929       112 Jun  2 22:48 llc.log
-rw-r--r--  1 dougo  1457291929  36182576 Jun  2 22:48 model.bc
-rw-r--r--  1 dougo  1457291929      1662 Jun  2 22:48 model.i
-rw-r--r--  1 dougo  1457291929      3883 Jun  2 22:48 model.i.h
-rw-r--r--  1 dougo  1457291929  29590676 Jun  2 22:48 model.o
-rw-r--r--  1 dougo  1457291929  36472416 Jun  2 22:48 model.opt.bc
-rw-r--r--  1 dougo  1457291929    101566 Jun  2 22:48 model.py
-rw-r--r--  1 dougo  1457291929    943857 Jun  2 22:48 modelPYTHON_wrap.cxx
-rw-r--r--  1 dougo  1457291929      1289 Jun  2 22:48 modelPYTHON_wrap.h
-rw-r--r--  1 dougo  1457291929        77 Jun  2 22:48 opt.log
```

At the python prompt inside of the 'host' folder, I get the following:
```
(py36) OSXFJK7L-719:host dougo$ python
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import model
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/dougo/code/ell/ELL/host/model.py", line 15, in <module>
    import _model
ModuleNotFoundError: No module named '_model'
```

Did I miss a step somewhere? Any tips to resolve? Thanks.

lovettchris, 2019-06-03:
assuming "build" folder contains the "built" version of this "host" project you should see a file named "_model.so" in there.  Then tutorial_helpers.py is used to load this module, notice it contains the following lines:

```python
# Find any child directory that matches the four deployment targets (pi3,
# pi3_64, aarch64, host) or begins with "model". For all these directories,
# add it and its platform-specific build directory to Python's import lookup
# path
SCRIPT_PATH = os.path.dirname(os.path.abspath(__file__))
SEARCH_DIRS = [os.path.join(SCRIPT_PATH, d) for d in os.listdir(SCRIPT_PATH) if
               os.path.isdir(d) and (
               d in ["pi3", "pi3_64", "aarch64", "host"] or
               d.startswith("model"))]
sys.path += SEARCH_DIRS
sys.path += [os.path.join(d, "build") for d in SEARCH_DIRS]
if platform.system() == "Windows":
    sys.path += [os.path.join(d, "build", "Release") for d in SEARCH_DIRS]
else:
    sys.path += [os.path.join(d, "build") for d in SEARCH_DIRS]
```
After this is done you can do "import model".

dougollerenshaw, 2019-06-03:
Thanks! That works, almost...

Now I get a 'Segmentation fault: 11' on import:
```
(py36) OSXFJK7L:ELL dougo$ python
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> import numpy as np
>>> import tutorial_helpers as helpers
>>> import model
Segmentation fault: 11

```
Here are a few relevant bing results:
https://stackoverflow.com/questions/12762944/segmentation-fault-11
https://stackoverflow.com/questions/10035541/what-causes-a-python-segmentation-fault
https://stackoverflow.com/questions/19531969/segmentation-fault-11-in-os-x

But it's not obvious to me what's actually going on. Any other tips? Is that worthy of a separate issue? Or have I made some other obvious mistake in the install/build process?

lovettchris, 2019-06-03:
That's not good.  Any chance you can chare your ELL model so I can repro this?

dougollerenshaw, 2019-06-03:
Sure. I just dropped the contents of the 'host' folder into a shared dropbox folder. It's here:

https://www.dropbox.com/sh/m24cnaipw7xfu6t/AAAbdynnlWTvM83vZMx7MTpua?dl=0
I installed/built on a macbook air with macOS Mojave 10.14.4

Is that sufficient to troubleshoot? I appreciate the help!

lovettchris, 2019-06-04:
Thanks, I see you provided the compiled bits, but not the *.ell file.  Can you share the .ell model you are compiling?

dougollerenshaw, 2019-06-04:
Here it is:
https://www.dropbox.com/sh/m24cnaipw7xfu6t/AAAbdynnlWTvM83vZMx7MTpua?dl=0

arachid1, 2020-06-12:
hey @dougollerenshaw did you have any luck figuring out where the seg fault comes from? I would appreciate your help.