microsoft/ELL #176: Unable to install OpenCV

bhakthil, 2018-09-01:
Hi,
I am having some trouble installing opencv onmy rpi3. it complains about missing dependency **openblas**.

(py34)pi@rpi3:~ $ conda install -y --channel microsoft-ell/label/stretch opencv
Fetching package metadata: ........
Solving package specifications:
Error: Could not find some dependencies for opencv: blas * openblas

You can search for this package on anaconda.org with
anaconda search -t conda blas * openblas

But, when I list down the packages, I clearly see openblas in the environment. how can I get this resolved?

(py34)pi@rpi3:~ $ conda list
packages in environment at /home/pi/miniconda3/envs/py34:
anaconda-client           1.0.2                    py34_0
clyent                    0.4.0                    py34_0
freetype                  2.5.2                         2
jpeg                      8d                            0
libgfortran               1.0                           0
libpng                    1.6.17                        0
libtiff                   4.0.2                         1
openblas                  0.2.14                        1
openssl                   1.0.1k                        1
pillow                    2.9.0                    py34_0
pip                       7.1.2                    py34_0
python                    3.4.3                         1
python-dateutil           2.4.2                    py34_0
pytz                      2015.4                   py34_0
pyyaml                    3.11                     py34_1
requests                  2.7.0                    py34_0
setuptools                18.1                     py34_0
six                       1.9.0                    py34_0
sqlite                    3.8.4.1                       1
wheel                     0.24.0                   py34_0
xz                        5.0.5                         0
yaml                      0.1.6                         0
zlib                      1.2.11                        0
(py34)pi@rpi3:~ $

Thanks
BL

lovettchris, 2018-09-03:
Perhaps the difference is that when I do the same I see OpenBlas 0.2.14 and not 0.2.14.1...

```
# packages in environment at /home/pi/miniconda3/envs/py34:
#
imutils                   0.4.6                     <pip>
jpeg                      8d                            0
libgfortran               1.0                           0
libpng                    1.6.17                        0
libtiff                   4.0.2                         1
numpy                     1.9.2                    py34_1
onnx                      1.2.2                     <pip>
openblas                  0.2.14                        1
opencv                    3.3.0                np19py34_0
openssl                   1.0.1k                        1
pip                       7.1.2                    py34_0
protobuf                  3.6.0                     <pip>
pyaudio                   0.2.11                    <pip>
python                    3.4.3                         1
setuptools                18.1                     py34_0
six                       1.11.0                    <pip>
sqlite                    3.8.4.1                       1
wheel                     0.24.0                   py34_0
xz                        5.0.5                         0
zlib                      1.2.8                         0
```

bhakthil, 2018-09-14:
I was able to build and run the samples outside of miniconda env. happy to provide the instructions if that would be helpful in the future.
BL

brunisgirl, 2018-09-30:
I get the same error, can you please provide the instructions for building it outside miniconda?

S.

phromo, 2019-01-10:
I'm getting this error on a new raspberry 3 model b+ running raspbian stretch. I'm following the instructions on https://microsoft.github.io/ELL/tutorials/Raspberry-Pi-setup/

If you managed to resolve this, can you please share?

bhakthil, 2019-01-10:
Hi,
I followed the below blog for installing OpenCV on virtualenv and completely ignored miniconda package manager. rest of the ELL tutorial still holds true for everything else.
https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/

harshmittal2210, 2019-01-10:
Hi,
I am also facing the same issue, I am getting one more message:
```
You may need to install the anaconda-client command line client with
              conda install anaconda-client
```

I even installed the anaconda-client still error is shown.

lovettchris, 2019-01-10:
There is also a pip installable open cv available here:
https://www.pyimagesearch.com/2018/09/19/pip-install-opencv/

also using virtualenv.  We are considering switching to virtualenv anyway...

phromo, 2019-01-12:
the pip package was really easy. Installed it directly to my root python. @lovettchris if you are revising your instructions, you might also want to consider pipenv (`pipenv shell`) which I've used in my recent projects. Collaborators find it intuitive that the state of the virtualenv is tied to a definition file (the `Pipfile`) instead of a procedural call of install commands.

lovettchris, 2019-01-25:
This should be fixed now, still using miniconda as per our setup instructions, just do this:
```
conda install -y -c microsoft-ell/label/stretch opencv
```
And you will now get 3 packages, including a newer version of openblas:
```
    openblas-0.3.6             |                1         7.0 MB
    numpy-1.9.2                |           py34_1         5.1 MB
    opencv-3.3.0               |       np19py34_0        33.2 MB
```

closed 2019-02-05 by lovettchris