microsoft/ELL #179: Building on Linux Mint 18.3 fails: recipe for 'Archiver.cpp.o' fails

stephansgit, 2018-11-19:
I am trying to build on **Linux Mint 18.3**.
`make` fails early in the process, related to `Archiver.cpp.o` on `TypeTraits.h`:

```
(py36) XXX@ideapad ~/ELL/build $ make
[  0%] Built target documentation
[  0%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Archiver.cpp.o
In file included from /home/XXX/ELL/libraries/utilities/include/TypeName.h:11:0,
                 from /home/XXX/ELL/libraries/utilities/include/Archiver.h:13,
                 from /home/XXX/ELL/libraries/utilities/src/Archiver.cpp:9:
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:88:38: error: ‘is_pointer_v’ is not a member of ‘std’
         template <typename T, bool = std::is_pointer_v<T>>
                                      ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:88:57: error: expected primary-expression before ‘>>’ token
         template <typename T, bool = std::is_pointer_v<T>>
                                                         ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:88:57: error: expected ‘>’ before ‘>>’ token
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:91:96: error: template argument 2 is invalid
             static constexpr size_t NumPointers = 1 + RemoveAllPointers<std::remove_pointer_t<T>>::NumPointers;
                                                                                                ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:92:76: error: template argument 2 is invalid
             using Type = typename RemoveAllPointers<std::remove_pointer_t<T>>::Type;
                                                                            ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:105:68: error: template argument 2 is invalid
     using RemoveAllPointersT = typename detail::RemoveAllPointers<T>::Type;
                                                                    ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:109:29: error: ‘ell::utilities::CountOfPointers’ declared as an ‘inline’ variable
     inline constexpr size_t CountOfPointers = detail::RemoveAllPointers<T>::NumPointers;
                             ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:109:74: error: template argument 2 is invalid
     inline constexpr size_t CountOfPointers = detail::RemoveAllPointers<T>::NumPointers;
                                                                          ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:113:54: error: parameter packs not expanded with ‘...’:
     struct VariantVisitor : T... { using T::operator()...; };
                                                      ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:113:54: note:         ‘T’
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:113:55: error: expected ‘;’ before ‘...’ token
     struct VariantVisitor : T... { using T::operator()...; };
                                                       ^
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:113:55: error: expected unqualified-id before ‘...’ token
/home/XXX/ELL/libraries/utilities/include/TypeTraits.h:117:49: error: expected constructor, destructor, or type conversion before ‘;’ token
     VariantVisitor(T...) -> VariantVisitor<T...>;
                                                 ^
libraries/utilities/CMakeFiles/utilities.dir/build.make:62: recipe for target 'libraries/utilities/CMakeFiles/utilities.dir/src/Archiver.cpp.o' failed
make[2]: *** [libraries/utilities/CMakeFiles/utilities.dir/src/Archiver.cpp.o] Error 1
CMakeFiles/Makefile2:2070: recipe for target 'libraries/utilities/CMakeFiles/utilities.dir/all' failed
make[1]: *** [libraries/utilities/CMakeFiles/utilities.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

Before that, `cmake ..` works, but gives some warnings:
```
CMake Deprecation Warning at /usr/local/share/cmake-3.11/Modules/UseSWIG.cmake:272 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:131 (swig_add_module)
  CMake/CommonInterfaces.cmake:163 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)

-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Writing: /home/XXX/ELL/build/config.json
-- Configuring done
CMake Warning at tools/utilities/debugCompiler/CMakeLists.txt:34 (add_executable):
  Cannot generate a safe runtime search path for target debugCompiler because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libffi.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/XXX/miniconda3/envs/py36/lib
    runtime library [libopenblas.so.0] in /usr/lib may be hidden by files in:
      /home/XXX/miniconda3/envs/py36/lib

  Some of these libraries may not be found correctly.

CMake Warning at tools/utilities/profile/CMakeLists.txt:28 (add_executable):
  Cannot generate a safe runtime search path for target profile because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libffi.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/XXX/miniconda3/envs/py36/lib
    runtime library [libopenblas.so.0] in /usr/lib may be hidden by files in:
      /home/XXX/miniconda3/envs/py36/lib

  Some of these libraries may not be found correctly.

CMake Warning at /usr/local/share/cmake-3.11/Modules/UseSWIG.cmake:323 (add_library):
  Cannot generate a safe runtime search path for target _ELL_python because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libffi.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/XXX/miniconda3/envs/py36/lib
    runtime library [libopenblas.so.0] in /usr/lib may be hidden by files in:
      /home/XXX/miniconda3/envs/py36/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.11/Modules/UseSWIG.cmake:273 (swig_add_library)
  CMake/CommonInterfaces.cmake:131 (swig_add_module)
  CMake/CommonInterfaces.cmake:163 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)

-- Generating done
-- Build files have been written to: /home/XXX/ELL/build
```

It looks like something very basic, but searching for the message does not reveal anything.
So I appreciate any support.

Thanks!

kernhanda, 2018-11-20:
Can you provide the entirety of the output from the initial `cmake ..` (the configuration) step?

stephansgit, 2018-11-20:
@kernhanda I've update my initial post with the full output of `cmake ..` - please see above.
Thanks for looking into it!

kernhanda, 2018-11-20:
Sorry, please delete `CMakeCache.txt` and run `cmake ..` again. In particular, I'm looking for your compiler with its full version.

stephansgit, 2018-11-20:
I edited my original post for missing information.
If it helps:

`cmake --version` gives `cmake version 3.11.3`;
`gcc-8 --version` gives `gcc-8 (Ubuntu 8.1.0-5ubuntu1~16.04) 8.1.0`

kernhanda, 2018-11-20:
Thanks. While you have GCC 8 installed, the cmake configuration step is using GCC 5.5.0.

> ```
> -- The C compiler identification is GNU 5.5.0                                                                                                                                                                                               
> -- The CXX compiler identification is GNU 5.5.0
> ```

Try deleting `CMakeCache.txt` and running the following `CC=gcc-8 CXX=g++-8 cmake ..` and then try building.

stephansgit, 2018-11-20:
@kernhanda, this solved the issue.
Thanks, really glad you took the time to look into it!

closed 2018-11-20 by stephansgit

VarshaSLalapura, 2019-03-28:
[error.txt](https://github.com/Microsoft/ELL/files/3016655/error.txt)
Hi @kernhanda  , am stuck at the same error

kernhanda, 2019-03-28:
> [error.txt](https://github.com/Microsoft/ELL/files/3016655/error.txt)
> Hi @kernhanda ,
> Am stuck at the same error
> Thanks in advance,
> Varsha

I don't see an error in your link. Just some CMake warnings that can be safely ignored. Do you see a failure when you build? If so, please open a new issue.

VarshaSLalapura, 2019-03-28:
I don't find the _model.pyd file in build\Release folder as mentioned in https://github.com/Microsoft/ELL/issues/101. 

This is my terminal: 

(py36) varshalalapura@hpcl011:~/Desktop/git-prjoects/ELL/ELL/tools/wrap/host$ python call_model.py 
Traceback (most recent call last):
  File "/home/varshalalapura/Desktop/git-prjoects/ELL/ELL/tools/wrap/host/model.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/varshalalapura/anaconda3/envs/py36/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named '_model'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "call_model.py", line 18, in <module>
    import model
  File "/home/varshalalapura/Desktop/git-prjoects/ELL/ELL/tools/wrap/host/model.py", line 17, in <module>
    _model = swig_import_helper()
  File "/home/varshalalapura/Desktop/git-prjoects/ELL/ELL/tools/wrap/host/model.py", line 16, in swig_import_helper
    return importlib.import_module('_model')
  File "/home/varshalalapura/anaconda3/envs/py36/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_model'

Do I open an issue freshly?
Thank you for your time and response,
Varsha

lovettchris, 2019-03-28:
Varsha I think your issue is unrelated to the original issue #179 here, can you open a new issue and provide more context on what you are trying to do?  Are you building some tutorial ?  Also Can you compare your call_model.py with [this one](https://github.com/Microsoft/ELL/blob/master/docs/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/call_model.py)?

VarshaSLalapura, 2019-03-29:
Yes Sir, am running a tutorial , image classification on RPi, have opened a new issue, #208