microsoft/ELL #180: Running tutorial fails with "Exception: You need to use the ModelWrapper directly..."

stephansgit, 2018-11-20:
Runnig the "get started" tutorial [here](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/) fails with the message 
`Exception: You need to use the ModelWrapper directly because this model is steppable, which means the input is provided by a callback method`

The full history is
```
>>> import cv2
>>> import numpy as np
>>> import tutorial_helpers as helpers
>>> import model
>>> input_shape = model.get_default_input_shape()
>>> output_shape = model.get_default_output_shape()
>>> print("Model input shape: [{0.rows}, {0.columns}, {0.channels}]".format(
...     input_shape))
Model input shape: [224, 224, 3]
>>> print("Model output shape: [{0.rows}, {0.columns}, {0.channels}]".format(
...     output_shape))
Model output shape: [1, 1, 1000]
>>> sample_image = cv2.imread("coffeemug.jpg")
>>> input_data = helpers.prepare_image_for_model(sample_image, input_shape.columns,
...                                              input_shape.rows)
>>> predictions = model.predict(input_data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "host/model.py", line 3381, in predict
    raise Exception("You need to use the ModelWrapper directly because this model is steppable, which means the input is provided by a callback method")
Exception: You need to use the ModelWrapper directly because this model is steppable, which means the input is provided by a callback method
``` 

I do see in `model.py` that this is a deliberate exception if `_model_wrapper.IsSteppable` - but what does that mean?
And how would I use the ModelWrapper directly?
(Absolute Python newbie)

Any help appreciated!

clovett, 2018-11-24:
oops, yeah, steppable api support has changed, so we need to fix the tutorial, but there's a work around. I pushed a new version of the model that is not "steppable" so the tutorial works again.  If you re-run this step it should work now:
```
curl --location -o model.ell.zip https://github.com/Microsoft/ELL-models/raw/master/models/ILSVRC2012/d_I224x224x3CMCMCMCMCMCMC1AS/d_I224x224x3CMCMCMCMCMCMC1AS.ell.zip
```

stephansgit, 2018-11-25:
Thanks; it works now!

closed 2018-11-25 by stephansgit