microsoft/ELL #182: Printing ELL nodes error on re purposing  pretrained classifier

vishnoitanuj, 2018-12-14:
input error: Error: couldn't read file: Failed to match field stride, instead found token 'extent'

lovettchris, 2019-01-07:
I just tried it an it works fine, can you try again with the latest bits?
```
(ell) d:\Temp\tutorials\retarget>%ELL_root%\build\bin\release\print.exe -imap pretrained.ell --includeNodeId --refineIterations 1
<id:1528> InputNode<float>(12288)
<id:1529> ConstantNode<float>()
<id:1530> ConstantNode<float>()
<id:1531> BroadcastLinearFunctionNode<float>(1528.output[0:12288], 1529.output[0:3], 1530.output[0:3])
<id:1532> ConstantNode<float>()
<id:1533> ConstantNode<float>()
<id:1534> BroadcastLinearFunctionNode<float>(1531.output[0:12288], 1532.output[0:3], 1533.output[0:0])
<id:1535> ConstantNode<float>()
<id:1536> ConstantNode<float>()
<id:1537> BroadcastLinearFunctionNode<float>(1534.output[0:12288], 1535.output[0:0], 1536.output[0:3])
<id:1538> ReorderDataNode<float>(1537.output[0:13068])
<id:1539> UnrolledConvolutionNode<float>(1538.output[0:13068])
<id:1540> ReorderDataNode<float>(1539.output[0:131072])
<id:1541> ConstantNode<float>()
<id:1542> ConstantNode<float>()
<id:1543> BroadcastLinearFunctionNode<float>(1540.output[0:131072], 1541.output[0:0], 1542.output[0:32])
<id:1544> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(1543.output[0:131072])
<id:1545> ConstantNode<float>()
<id:1546> ConstantNode<float>()
<id:1547> BroadcastLinearFunctionNode<float>(1544.output[0:131072], 1545.output[0:32], 1546.output[0:32])
<id:1548> ConstantNode<float>()
<id:1549> ConstantNode<float>()
<id:1550> BroadcastLinearFunctionNode<float>(1547.output[0:131072], 1548.output[0:32], 1549.output[0:0])
<id:1551> ConstantNode<float>()
<id:1552> ConstantNode<float>()
<id:1553> BroadcastLinearFunctionNode<float>(1550.output[0:131072], 1551.output[0:0], 1552.output[0:32])
<id:1554> ReorderDataNode<float>(1553.output[0:139392])
<id:1555> UnrolledConvolutionNode<float>(1554.output[0:139392])
<id:1556> ReorderDataNode<float>(1555.output[0:131072])
<id:1557> ConstantNode<float>()
<id:1558> ConstantNode<float>()
<id:1559> BroadcastLinearFunctionNode<float>(1556.output[0:131072], 1557.output[0:0], 1558.output[0:32])
<id:1560> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(1559.output[0:131072])
<id:1561> PoolingLayerNode<float,MaxPoolingFunction>(1560.output[0:131072])
    PoolingLayer<float,MaxPoolingFunction>(shape=[64,64,32]->[32,32,32], function=maxpooling, stride=2, size=2)
<id:1562> ConstantNode<float>()
<id:1563> ConstantNode<float>()
<id:1564> BroadcastLinearFunctionNode<float>(1561.output[0:32768], 1562.output[0:32], 1563.output[0:32])
<id:1565> ConstantNode<float>()
<id:1566> ConstantNode<float>()
<id:1567> BroadcastLinearFunctionNode<float>(1564.output[0:32768], 1565.output[0:32], 1566.output[0:0])
<id:1568> ConstantNode<float>()
<id:1569> ConstantNode<float>()
<id:1570> BroadcastLinearFunctionNode<float>(1567.output[0:32768], 1568.output[0:0], 1569.output[0:32])
<id:1571> ReorderDataNode<float>(1570.output[0:36992])
<id:1572> UnrolledConvolutionNode<float>(1571.output[0:36992])
<id:1573> ReorderDataNode<float>(1572.output[0:65536])
<id:1574> ConstantNode<float>()
<id:1575> ConstantNode<float>()
<id:1576> BroadcastLinearFunctionNode<float>(1573.output[0:65536], 1574.output[0:0], 1575.output[0:64])
<id:1577> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(1576.output[0:65536])
<id:1578> ConstantNode<float>()
<id:1579> ConstantNode<float>()
<id:1580> BroadcastLinearFunctionNode<float>(1577.output[0:65536], 1578.output[0:64], 1579.output[0:64])
<id:1581> ConstantNode<float>()
<id:1582> ConstantNode<float>()
<id:1583> BroadcastLinearFunctionNode<float>(1580.output[0:65536], 1581.output[0:64], 1582.output[0:0])
<id:1584> ConstantNode<float>()
<id:1585> ConstantNode<float>()
<id:1586> BroadcastLinearFunctionNode<float>(1583.output[0:65536], 1584.output[0:0], 1585.output[0:64])
<id:1587> ReorderDataNode<float>(1586.output[0:73984])
<id:1588> UnrolledConvolutionNode<float>(1587.output[0:73984])
<id:1589> ReorderDataNode<float>(1588.output[0:65536])
<id:1590> ConstantNode<float>()
<id:1591> ConstantNode<float>()
<id:1592> BroadcastLinearFunctionNode<float>(1589.output[0:65536], 1590.output[0:0], 1591.output[0:64])
<id:1593> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(1592.output[0:65536])
[... 61 lines cut ...]
<id:1652> ConstantNode<float>()
<id:1653> BroadcastLinearFunctionNode<float>(1650.output[0:4096], 1651.output[0:0], 1652.output[0:256])
<id:1654> ReorderDataNode<float>(1653.output[0:9216])
<id:1655> UnrolledConvolutionNode<float>(1654.output[0:9216])
<id:1656> ReorderDataNode<float>(1655.output[0:8192])
<id:1657> ConstantNode<float>()
<id:1658> ConstantNode<float>()
<id:1659> BroadcastLinearFunctionNode<float>(1656.output[0:8192], 1657.output[0:0], 1658.output[0:512])
<id:1660> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(1659.output[0:8192])
<id:1661> PoolingLayerNode<float,MaxPoolingFunction>(1660.output[0:8192])
    PoolingLayer<float,MaxPoolingFunction>(shape=[4,4,512]->[2,2,512], function=maxpooling, stride=2, size=2)
<id:1662> ReorderDataNode<float>(1661.output[0:2048])
<id:1663> UnrolledConvolutionNode<float>(1662.output[0:2048])
<id:1664> ReorderDataNode<float>(1663.output[0:4000])
<id:1665> ConstantNode<float>()
<id:1666> ConstantNode<float>()
<id:1667> BroadcastLinearFunctionNode<float>(1664.output[0:4000], 1665.output[0:0], 1666.output[0:1000])
<id:1668> BroadcastUnaryFunctionNode<float,ReLUActivationFunction<float>>(1667.output[0:4000])
<id:1669> PoolingLayerNode<float,MeanPoolingFunction>(1668.output[0:4000])
    PoolingLayer<float,MeanPoolingFunction>(shape=[2,2,1000]->[1,1,1000], function=meanpooling, stride=1, size=2)
<id:1670> SoftmaxLayerNode<float>(1669.output[0:1000])
    SoftmaxLayer<float>(shape=[1,1,1000]->[1,1,1000])
<id:1671> OutputNode<float>(1670.output[0:1000])
```
```

kernhanda, 2019-01-10:
Fixed with release [v2.3.7](https://github.com/Microsoft/ELL/releases/tag/v2.3.7).

closed 2019-01-10 by kernhanda