microsoft/ELL #184: Browser WASM and ELL

grahamehorner, 2018-12-18:
At present it is not possible to use ML.NET with in s #Blazor client application, give that a browser WASM application is in essence a VM with limited resources, it would IMHO be a great for ELL to target client browser possibly emitting Typescript or a C# ELL predictor that can be included I to a C# #Blazor application

lovettchris, 2018-12-18:
Agreed :-) This may already be possible if you can find an LLVM bitcode to JavaScript generator, like say [emscripten](https://github.com/kripken/emscripten).  The ELL compiler has a "-bitcode" output option already.  But I haven't tried whether this actually works... probably also need to compile the ELL model with "--blas false" so that your bitcode has no openblas dependency.

grahamehorner, 2018-12-18:
So something like mono-llvm ?

grahamehorner, 2018-12-18:
Is it at all possible to swap out the llvm implementation used by ELL with something like https://github.com/Sonicadvance1/llvm-xtensa to attempt generation of code for ESP32 ?