microsoft/ELL #185: VS 2019

grahamehorner, 2018-12-20:
allow build using VS2019 preview

clovett, 2018-12-20:
Have you tried loading the CMakeLists.txt directly (VS now supports cmake directly).  If that doesn't work you can probably find the cmake generator manually using the version of cmake that comes with VS.  If you type "cmake -G" you will see the list of supported VS versions under Generators, I'm guessing it will be named "Visual Studio 16 2019 Win64" and then you might be able to do this in the ELL folder :
```
mkdir build
cd build
cmake -G "Visual Studio 16 2019 Win64" ..
cmake --build . --config Release
```
Does this work for you?

clovett, 2019-01-07:
When you rebuild y our VM let's go with VS 2017 this time, then it should be good.

lovettchris, 2019-04-29:
closing due to inactivity, please reopen if you want this resolved, thanks

closed 2019-04-29 by lovettchris