microsoft/ELL #186: Q: target devices/languages

grahamehorner, 2018-12-20:
any plans to support Arduino devices with samples ?

clovett, 2018-12-20:
The Azure IOT devkit used in [this demo ](https://github.com/Microsoft/ELL/wiki/Keyword-Spotting-on-MXCHIP) is actually running Arduino.  The thing is "Arduino devices" is ambiguous, Arduino firmware runs on [lots of different types of devices](https://en.wikipedia.org/wiki/List_of_Arduino_boards_and_compatible_systems) these days.  If your device contains an ARM chip then ELL can compile models for that chip.  If it is an ATMEL chip then the current answer is we haven't tried it, so probably not.  It is on our list of things to support, but is not actively being worked on.  ELL uses LLVM and there is an [AVR LLVM backend](http://llvm.org/viewvc/llvm-project/llvm/trunk/lib/Target/AVR/) so theoretically it should not be too hard to hook it up.  ELL can generate bitcode, so you should be able to find an LLVM compiler that can compile that bitcode to AVR.  We would love for someone in the community to try it and create a Pull Request.  Having said that don't expect great performance from a typical model unless your chip has good floating point vectorization.  You may need to quantize your models to integer values to get any kind of performance from atmel chips, but ELL does not yet fully support integerized models, so that would be another chunk of work.  The internal ELL nodes are templatized on element type which we use to support float and double, so it should not be too hard to support int as well, but there will be work required in the math library to do integer computation in a way that doesn't loose too much precision.

grahamehorner, 2018-12-21:
@clovett thanks it’s really interesting work you guys are doing and it would be good to understand the longer term roadmap and/or vision of the chips/languages you are thinking of supporting. Keep up the great work loving the ideas 💡

clovett, 2018-12-31:
Cool, thanks for the encouragement.  I'll pass along your request for longer term roadmap.

closed 2019-01-22 by lovettchris