microsoft/ELL #187: Q: support for IL2CPP and/or CoreRT

grahamehorner, 2019-01-02:


lovettchris, 2019-01-07:
No.  But I see no reason why these things couldn't already work with the compiled neural network output from ELL:

[image]

and
[image]

grahamehorner, 2019-01-07:
I was thinking something more like 

neural network -> ELL -> C# -> CoreRT -> C++ ?
neural network -> ELL -> object + .h -> ?

as then ELL could build upon the CoreRT efforts and perhaps be deployed onto a platform that is not currently supported; but becomes a supported CoreRT platform ?

all just thinking aloud with ideas for the future; loving the fact ML/AI is running on the edge and the work you guys are doing :D  +1000

clovett, 2019-01-07:
Well you might be looking for an ELL to C++ option, this would then give you more platform options (any platform that can compile that C++ code), but this is not currently supported.  But it is on our list of things to look into.

grahamehorner, 2019-01-07:
@clover cool would love to contribute as when if you guys start this or believe the effort is worth the time

lovettchris, 2019-01-08:
Sure thing, love to get some help.  In the meantime if you use IL2CPP or CoreRT to native code there it should already by possible to link the ELL generated object code into an existing native application.  I'd be happy to help get that working if you want to try something along those lines.

closed 2019-01-22 by lovettchris