microsoft/ELL #188: cmake

rverm44820, 2019-01-06:
ii have going threw the windows install.txt. this is where im stuck. do i install the cmake program in the ell folder?

(py36) C:\Users\Rob\build>cmake -G "Visual Studio 15 2017 Win64" ..
'cmake' is not recognized as an internal or external command,
operable program or batch file.

(py36) C:\Users\Rob\build>cmake -G
'cmake' is not recognized as an internal or external command,
operable program or batch file.

lovettchris, 2019-01-07:
Assuming you have [Visual Studio 2017](https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=Community&rel=15) installed you should be able to open a "Developer Command Prompt" for Visual studio which looks like this:

[image]

and then cmake will be available (it lives here: C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe)

rverm44820, 2019-01-07:
Is visual code studio the same program?

grahamehorner, 2019-01-07:
@rverm44820 No, Visual Studio & Visual Studio Code are NOT the same, Visual Studio is a commercial IDE; yet is offered for free as a Community Edition for Windows & iOS it has a lot more functionality the VS Code.  VS Code is a free cross platform IDE that allows development on Windows, iOS and a number of Linux operating systems.   

I would recommend using the Visual Studio 2017 Community Edition for this repo in the first instance.

lovettchris, 2019-01-10:
Right.  VS Code is great for python development, but you need full Visual Studio to get the C++ compiler that is needed to build ELL.

closed 2019-01-10 by lovettchris