microsoft/ELL #190: cmake error while deploying on windows 10

harshmittal2210, 2019-01-07:
While following the steps in tutorial at [page](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/)

The last Step till which no error were shown gave the following result, as mention in the tutorials
```
compiling model...
generating python interfaces for model in host
running opt ...
running llc ...
success, now you can build the 'host' folder
```
the next step
```
> cmake -G "Visual Studio 14 2015 Win64" .. && cmake --build . --config release && cd ..\..
```

Gave the following result:
```
-- Selecting Windows SDK version  to target Windows 10.0.17134.
CMake Error at CMakeLists.txt:34 (project):
  Failed to run MSBuild command:

    MSBuild.exe

  to get the value of VCTargetsPath:

    The system cannot find the file specified

-- Configuring incomplete, errors occurred!
See also "<Dir_loc>/host/build/CMakeFiles/CMakeOutput.log".
```

lovettchris, 2019-01-08:
I wonder if your Visual Studio installation is missing the C++ feature?  This is what I have installed:

[image]

harshmittal2210, 2019-01-08:
I have install C++ features as shown in the your screenshot @lovettchris , still I am getting this error

clovett, 2019-01-08:
Ah, whenever you see a message like that simply delete the "build" folder and start over.