microsoft/ELL #195: C++ tutorial build does not seem to complete

mrohera, 2019-02-04:
Following the instructions here: https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-in-cpp/, I am seeing that the build is essentially stalled and I have to CTRL+C to kill the build. Could you kindly suggest what am I missing?

```
python /home/marohera/git/ell/ELL/tools/wrap/wrap.py --model_file model.ell --language cpp --outdir model              
copy "/home/marohera/git/ell/ELL/CMake/OpenBLASSetup.cmake" "model/OpenBLASSetup.cmake"
compiling model...
/home/marohera/git/ell/ELL/build/bin/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od model --fuseLinearOps True --optimizeReorderDataNodes True --header --blas true --optimize true
running opt ...
/usr/lib/llvm-6.0/bin/opt model/model.bc -o model/model.opt.bc -O3
running llc ...
/usr/lib/llvm-6.0/bin/llc model/model.opt.bc -o model/model.o -O3 -filetype=obj -relocation-model=pic
```

lovettchris, 2019-02-05:
Try running this to produce the required model.o object file:
```
d:/git/ELL/ELL/build/bin/release/compile -imap model.ell -cfn Predict -cmn model --target host -od model --fuseLinearOps True --optimizeReorderDataNodes True --header --blas true --optimize true --objectCode
```

mrohera, 2019-02-05:
Thanks for getting back! After following the tutorial I created the build dir and ran cmake. I do not have the compile executable as per your comment above. If you have a writeup to generate a model.o that is perfect and exactly what I am after.

cjacobs, 2019-02-05:
Hi,
Is the `compile` executable perhaps in `/homne/marohera/git/ell/ELL/build/bin`? (without the 'release')?

mrohera, 2019-02-05:
Found it under /ELL/build/bin sorry it was my bad, I issued the command and looks like it worked. Thanks Chris.
```
ls -lr model 
total 100140
-rw-r--r-- 1 marohera marohera       66 Feb  4 14:31 opt.log
-rw-r--r-- 1 marohera marohera     7813 Feb  4 14:30 OpenBLASSetup.cmake
-rw-r--r-- 1 marohera marohera 36665152 Feb  4 14:31 model.opt.bc
-rw-r--r-- 1 marohera marohera 29588496 Feb  5 11:06 model.o
-rw-r--r-- 1 marohera marohera     3816 Feb  5 11:06 model.h
-rw-r--r-- 1 marohera marohera 36254288 Feb  4 14:31 model.bc
-rw-r--r-- 1 marohera marohera      101 Feb  4 14:31 llc.log
drwxr-xr-x 2 marohera marohera     4096 Feb  3 17:53 include
-rw-r--r-- 1 marohera marohera      208 Feb  4 14:30 compile.log
```

mrohera, 2019-02-05:
Just FYI here is the output of /proc/cpuinfo/
```
cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 58
model name	: Intel(R) Core(TM) i7-3667U CPU @ 2.00GHz
```
This is the Intel's product page for this CPU: https://ark.intel.com/products/64898/Intel-Core-i7-3667U-Processor-4M-Cache-up-to-3-20-GHz-

lovettchris, 2019-02-05:
Oh, and be sure to add "--target pi3" to the compile command line if you plan to run that model on your raspberry pi.