microsoft/ELL #200: wrap.py: WrapException: <class 'buildtools.EllBuildToolsRunException'>

jnbli, 2019-02-25:
I've put model.ell and categories.txt in a the same folder and run wrap.py, and I am getting an exception.

python3 /Users/jonathanli/Applications/squirrel-tracker/ELL/tools/wrap/wrap.py --model_file model.ell --language python --target host

copy "/Users/jonathanli/Applications/squirrel-tracker/ELL/CMake/OpenBLASSetup.cmake" "host/OpenBLASSetup.cmake"
compiling model...
/Users/jonathanli/Applications/squirrel-tracker/ELL/build/bin/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas true --optimize true
### WrapException: <class 'buildtools.EllBuildToolsRunException'>: /Users/jonathanli/Applications/squirrel-tracker/ELL/build/bin/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas true --optimize true

lovettchris, 2019-02-25:
Can you run just the "compile" part of this command again with "--verbose" and report the output you see?
```
/Users/jonathanli/Applications/squirrel-tracker/ELL/build/bin/compile -imap model.ell -cfn Predict -cmn model --bitcode --target host -od host --fuseLinearOps True --optimizeReorderDataNodes True --swig --blas true --optimize true --verbose
```

lovettchris, 2019-04-29:
Please try the latest bits.

closed 2019-04-29 by lovettchris