microsoft/ELL #201: OpenBLAS on Ryzen 2500u

Ry7en, 2019-03-11:
Following the instructions on the "Building a 64-bit version of the Embedded Learning Library (ELL) on Windows" page I reached an impasse at the "Invoke CMake as follows" instruction ("-- Unknown processor, disabling OpenBLAS"). A google search led me to believe that this is caused by my laptops processor, an AMD 2500u, not being included in the CMake\OpenBLASSetup.cmake script, but I couldn't find the instructions, suggested on a thread on the issues page here, required to build OpenBLAS on my computer. Any help would be appreciated.

lovettchris, 2019-03-11:
clone [https://github.com/xianyi/OpenBLAS](https://github.com/xianyi/OpenBLAS), then:
```
mkdir build
cd build
cmake -G "Visual Studio 15 2017 Win64" ..
cmake --build . --config Release
```
This will give you an OpenBlas build that works for your computer.  The output of this build is in OpenBLAS\build\lib\release where you will see libopenblas.dll.  Then run this from an Admin command prompt:
```
cmake --build . --config Release --target INSTALL
```
This will put the installed bits in "C:/Program Files/OpenBLAS" and the next step is helping the ELL build find this build of OpenBlas.  The easiest way might be something like this before the FindPackage call:
```
set(BLAS_DLL_DIR "C:/Program Files/OpenBLAS/bin")
set(BLAS_INCLUDE_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/include/)
set(BLAS_LIB_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/lib/)
```

lovettchris, 2019-03-11:
It is a  bit more work to get OpenBlasSetup.cmake to find this version, see attached updates.

[OpenBLASSetup.zip](https://github.com/Microsoft/ELL/files/2953929/OpenBLASSetup.zip)

Ry7en, 2019-03-11:
Thank you very much for your prompt reply; after running "cmake -G "Visual Studio 15 2017 Win64" .." the output contained a number of warnings, namely:
a "CMake Deprecation Warning";
multiple  "Policy CMP0054 is not set" warnings;
a "Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)" warning.

```
(py36) C:\Users\HP\OpenBLAS\build>cmake -G "Visual Studio 15 2017 Win64" ..
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- The C compiler identification is MSVC 19.16.27027.1
-- The ASM compiler identification is MSVC
-- Found assembler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Warning at CMakeLists.txt:59 (message):
  CMake support is experimental.  It does not yet support all build options
  and may not produce the same Makefiles that OpenBLAS ships with.

-- GEMM multithread threshold set to 4.
-- Multi-threading enabled with 8 threads.
CMake Deprecation Warning at C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.12/Modules/CMakeForceCompiler.cmake:97 (message):
  The CMAKE_FORCE_Fortran_COMPILER macro is deprecated.  Instead just set
  CMAKE_Fortran_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/f_check.cmake:27 (CMAKE_FORCE_Fortran_COMPILER)
  cmake/prebuild.cmake:84 (include)
  cmake/system.cmake:141 (include)
  CMakeLists.txt:62 (include)

CMake Warning (dev) at cmake/prebuild.cmake:298 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  cmake/system.cmake:141 (include)
  CMakeLists.txt:62 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

MSVC
-- Running getarch
-- GETARCH results:
CORE=generic
LIBCORE=generic
NUM_CORES=8
GENERIC=1
L1_DATA_SIZE=32768
L1_DATA_LINESIZE=128
L2_SIZE=512488
L2_LINESIZE=128
DTB_DEFAULT_ENTRIES=128
DTB_SIZE=4096
L2_ASSOCIATIVE=8
MAKE += -j 8

-- Building Single Precision
-- Building Double Precision
-- Building Complex Precision
-- Building Double Complex Precision
-- Reading vars from C:/Users/HP/OpenBLAS/kernel/x86_64/KERNEL...
-- Reading vars from C:/Users/HP/OpenBLAS/kernel/x86_64/KERNEL.generic...
CMake Warning (dev) at kernel/CMakeLists.txt:124 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "GENERIC" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  kernel/CMakeLists.txt:546 (build_core)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
CMake Warning (dev) at utest/CMakeLists.txt:4 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating openblas_config.h in include/openblas
-- Generating f77blas.h in include/openblas
-- Generating cblas.h in include/openblas
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/HP/OpenBLAS/build
```

In my previous attempt to resolve the problem of getting OpenBLAS to work I tried following the instructions for a "Native (MSVC) ABI" build (https://github.com/xianyi/OpenBLAS/wiki/How-to-use-OpenBLAS-in-Microsoft-Visual-Studio), and I recall that some packages installed previously during the ELL installation were up and down-graded. Perhaps this is causing the PkgConfig warning?

lovettchris, 2019-03-11:
You can safely ignore all those warnings, go ahead with the build step:
```
cmake --build . --config Release
```

Ry7en, 2019-03-11:
The build step has completed with no errors, but 515 warnings, which mostly seem to be code C4244:

`conversion from 'BLASLONG' to 'int', possible loss of data`

lovettchris, 2019-03-11:
Great, now for the install step from an admin command prompt:
```
cmake --build . --config Release --target INSTALL
```

Ry7en, 2019-03-11:
The install proceeded with no errors, but for the next step should I skip the last instruction in your first post, and replace the OpenBLASSetup.cmake in my ELL/CMake folder with the one you attached in your second post?

lovettchris, 2019-03-11:
yes, replace OpenBLASSetup.cmake in your ELL/Cmake with the one I attached earlier, then delete any ELL/build folder you may have created earlier, then follow the ELL build instrucrtions.

Ry7en, 2019-03-12:
After following the "Building ELL" section of the installation guide and running "cmake -G "Visual Studio 15 2017 Win64" ..", the following was output, including the "-- Unknown processor, disabling OpenBLAS" warning :

```
(py36) C:\Users\HP\ELL\build>cmake -G "Visual Studio 15 2017 Win64" ..
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- The C compiler identification is MSVC 19.16.27027.1
-- The CXX compiler identification is MSVC 19.16.27027.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ELL version 2.4.0
-- Configuring tests to use TEST_MODELS_REPO at: https://github.com/Microsoft/ell-test-models
-- PYTHONINTERP_FOUND=FALSE
-- PYTHON_EXECUTABLE=C:/Users/HP/Miniconda3/envs/py36/python.exe
-- PYTHON_VERSION_STRING=3.6.8
-- PYTHON_VERSION_MAJOR=3
-- PYTHONLIBS_FOUND=FALSE
-- PYTHON_LIBRARIES=C:/Users/HP/Miniconda3/envs/py36/libs/python36.lib
-- PYTHON_INCLUDE_PATH=C:/Users/HP/Miniconda3/envs/py36/include
-- PYTHON_INCLUDE_DIRS=C:/Users/HP/Miniconda3/envs/py36/include
-- PYTHON_DEBUG_LIBRARIES=PYTHON_DEBUG_LIBRARY-NOTFOUND
-- PYTHONLIBS_VERSION_STRING=3.6.8
-- PYTHONINTERP_FOUND=TRUE
-- PYTHON_EXECUTABLE=C:/Users/HP/Miniconda3/envs/py36/python.exe
-- PYTHON_VERSION_STRING=3.6.8
-- PYTHON_VERSION_MAJOR=3
-- PYTHONLIBS_FOUND=TRUE
-- PYTHON_LIBRARIES=C:/Users/HP/Miniconda3/envs/py36/libs/python36.lib
-- PYTHON_INCLUDE_PATH=C:/Users/HP/Miniconda3/envs/py36/include
-- PYTHON_INCLUDE_DIRS=C:/Users/HP/Miniconda3/envs/py36/include
-- PYTHON_DEBUG_LIBRARIES=PYTHON_DEBUG_LIBRARY-NOTFOUND
-- PYTHONLIBS_VERSION_STRING=3.6.8
CMake Warning at CMake/Flake8.cmake:7 (message):
  ### could not find 'flake8', please run 'pip install flake8'
Call Stack (most recent call first):
  CMakeLists.txt:86 (include)

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Processor family: 23, model: 17
CMake Error at CMake/OpenBLASSetup.cmake:171 (get_processor_mapping):
  get_processor_mapping Macro invoked with incorrect arguments for macro
  named: get_processor_mapping
Call Stack (most recent call first):
  CMakeLists.txt:145 (include)

-- Unknown processor, disabling OpenBLAS
-- BLAS library not found
-- Found LLVM 6.0.1
-- Using LLVMConfig.cmake in: C:/Users/HP/ELL/external/LLVMNativeWindowsLibs.x64.6.0.1/llvm-6.0/lib/cmake/llvm
-- Found SWIG: C:/Users/HP/ELL/external/swigwintools.3.0.12/tools/swigwin-3.0.12/swig.exe (found suitable version "3.0.12", minimum required is "3.0.12")
-- Found SWIG_EXECUTABLE=C:/Users/HP/ELL/external/swigwintools.3.0.12/tools/swigwin-3.0.12/swig.exe
-- Skipping pitest because one of RPI_CLUSTER, RPI_PASSWORD, RPI_KEY, TEST_MODELS_REPO is missing
-- Creating wrappers for python
CMake Deprecation Warning at C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.12/Modules/UseSWIG.cmake:492 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:129 (swig_add_module)
  CMake/CommonInterfaces.cmake:160 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)

-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Writing: C:/Users/HP/ELL/build/config.json
-- Configuring incomplete, errors occurred!
See also "C:/Users/HP/ELL/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/HP/ELL/build/CMakeFiles/CMakeError.log".
```

lovettchris, 2019-03-12:
```
CMake Error at CMake/OpenBLASSetup.cmake:171 (get_processor_mapping):
  get_processor_mapping Macro invoked with incorrect arguments for macro
  named: get_processor_mapping
```
Huh, what version of cmake are you using?  This is the one I'm using:
```
C:\Program Files\CMake\bin\cmake.exe
C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe

(ell) c:\git\ELL\ELL>cmake --version
cmake version 3.10.0
```

Ry7en, 2019-03-12:
I get:

```
C:\Users\HP\ELL\CMake>cmake --version
cmake version 3.12.18081601-MSVC_2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
I believe that CMake may have been updated during my aforementioned attempt to follow the "Native (MSVC) ABI" OpenBLAS build tutorial.

lovettchris, 2019-03-12:
Hmm, I just tried my modified OpenBLASSetup.cmake on another machine with the exact same version of cmake (cmake version 3.12.18081601-MSVC_2) and it works fine... can you run the following on your machine and report the output?
```
(ell) d:\temp\ELL\ELL\CMake>set proc
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 79 Stepping 1, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=4f01
```
I suspect we have a bug in our get_processor_mapping macro...

Ry7en, 2019-03-12:
In the first example I get:

```
C:\Users\HP\ELL\CMake>set proc
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=AMD64 Family 23 Model 17 Stepping 0, AuthenticAMD
PROCESSOR_LEVEL=23
PROCESSOR_REVISION=1100
```
Whilst the second has:

```

C:\Users\HP\ELL\CMake>reg query HKEY_LOCAL_MACHINE\Hardware\Description\System\CentralProcessor\0

HKEY_LOCAL_MACHINE\Hardware\Description\System\CentralProcessor\0
    Component Information    REG_BINARY    00000000000000000000000000000000
    Identifier    REG_SZ    AMD64 Family 23 Model 17 Stepping 0
    Configuration Data    REG_FULL_RESOURCE_DESCRIPTOR    FFFFFFFFFFFFFFFF0000000000000000
    ProcessorNameString    REG_SZ    AMD Ryzen 5 PRO 2500U w/ Radeon Vega Mobile Gfx
    VendorIdentifier    REG_SZ    AuthenticAMD
    FeatureSet    REG_DWORD    0x3c3b3dff
    ~MHz    REG_DWORD    0x7cc
    Update Revision    REG_BINARY    0B10100800000000
    Update Status    REG_DWORD    0x1
    Previous Update Revision    REG_BINARY    0B10100800000000
    Platform Specific Field1    REG_DWORD    0x810100b

```

lovettchris, 2019-03-12:
Oh, hang on, the OpenBLASSetup.cmake shouldn't be getting to line 171.  It should be finding your OpenBlas installation in c:\program files\openblas… but this output means it did not find that:
```
-- BLAS library not found
```
So we need to debug why the new lines I added 108-121 didn't succeed in finding your openblas install...  running this line and attaching a zipped version of the trace.log file will help with that:
```
cmake -G "Visual Studio 15 2017 Win64" .. --trace > trace.log 2>&1
```

Ry7en, 2019-03-12:
I hope this is what was required:

[trace.zip](https://github.com/Microsoft/ELL/files/2958712/trace.zip)

lovettchris, 2019-03-12:
Yes, thanks that was helpful.  Looks like OpenBLASSetup.cmake is not finding your OpenBlas install.  You should see this output in the trace.log file:
```
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(93):  if(BLAS_FOUND )
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(105):  else()
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(106):  if(WIN32 )
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(109):  if(${CMAKE_SYSTEM_PROCESSOR} STREQUAL AMD64 )
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(110):  set(BLAS_PACKAGE_DIR $ENV{ProgramW6432}/OpenBlas )
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(115):  if(EXISTS ${BLAS_PACKAGE_DIR}/lib/libopenblas.lib )
D:/temp/ELL/ELL/CMake/OpenBLASSetup.cmake(116):  message(STATUS Found OpenBlas in ${BLAS_PACKAGE_DIR} )
-- Found OpenBlas in C:\Program Files/OpenBlas
```
But that last line is missing from your trace.  Can you check your OpenBlas INSTALL step worked properly and put open blas in this location "C:\Program Files\OpenBlas" ?

Ry7en, 2019-03-12:
As far as I can tell, the install step was successful - "C:\Program Files\OpenBLAS" is present and contains three folders; "include", "lib" and "share".

lovettchris, 2019-03-12:
hmmm, ok, perhaps on an AMD machine the CMAKE_SYSTEM_PROCESSOR is set to something different, can you print out the result of message(STATUS CMAKE_SYSTEM_PROCESSOR=${CMAKE_SYSTEM_PROCESSOR}) ?  Just add this to a tiny cmake file and run the cmake generator step (cmake -G "Visual Studio 15 2017 Win64").  You should see something like this in the cmake output:
```
-- CMAKE_SYSTEM_PROCESSOR=AMD64
```

Ry7en, 2019-03-12:
I don't think I've interpreted your instructions correctly, but nevertheless I added "message(STATUS CMAKE_SYSTEM_PROCESSOR=${CMAKE_SYSTEM_PROCESSOR})" to OpenBLASSetup.cmake, ran "cmake -G "Visual Studio 15 2017 Win64" .." (I had to add the two periods at the end otherwise I would get a CMakeLists.txt missing error) which included in its output:

`-- CMAKE_SYSTEM_PROCESSOR=AMD64`

instead of the lines previously output
 ```

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE

```

lovettchris, 2019-03-13:
Sorry I wasn't clear.  I didn't mean for you to edit the ELL cmake files.  Instead create a new empty folder named "foo" someplace, add one text file named "CMakeLists.txt", containing this:
```
message(STATUS CMAKE_SYSTEM_PROCESSOR=${CMAKE_SYSTEM_PROCESSOR})
```
then in this folder run this command:
```
cmake -G "Visual Studio 15 2017 Win64"
```
and send the output that you see.  If this works, then we can build up from there to this and figure out why it isn't working:
```
# This is the default location for a manually installed OpenBlas build.
if(${CMAKE_SYSTEM_PROCESSOR} STREQUAL "AMD64")
    set(BLAS_PACKAGE_DIR $ENV{ProgramW6432}/OpenBlas)
else()
    set(BLAS_PACKAGE_DIR $ENV{ProgramFiles}/OpenBlas)
endif()

if(EXISTS ${BLAS_PACKAGE_DIR}/lib/libopenblas.lib)
    message(STATUS "Found OpenBlas in ${BLAS_PACKAGE_DIR}")
    set(BLAS_DLLS libopenblas.dll)
    set(BLAS_DLL_DIR ${BLAS_PACKAGE_DIR}/bin)
    set(BLAS_INCLUDE_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/include/)
    set(BLAS_LIB_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/lib/)
    set(BLAS_FOUND TRUE)
endif()
```

Ry7en, 2019-03-13:
Thank you for persevering with me, I've undone the changes to OpenBLASSetup.cmake; the output from 

`cmake -G "Visual Studio 15 2017 Win64"`

is:
```

C:\Users\HP\Documents\foo>cmake -G "Visual Studio 15 2017 Win64"
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- The C compiler identification is MSVC 19.16.27027.1
-- The CXX compiler identification is MSVC 19.16.27027.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM_PROCESSOR=AMD64
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/HP/Documents/foo
```

lovettchris, 2019-03-13:
Ok, great the important thing here is we see this output:
```
-- CMAKE_SYSTEM_PROCESSOR=AMD64
```
This is good, now please modify your little test CMakeLists.txt so it contains this:
```
# This is the default location for a manually installed OpenBlas build.
message(STATUS "ProgramW6432=[$ENV{ProgramW6432}]")
if(${CMAKE_SYSTEM_PROCESSOR} STREQUAL "AMD64")
    set(BLAS_PACKAGE_DIR $ENV{ProgramW6432}/OpenBlas)
else()
    set(BLAS_PACKAGE_DIR $ENV{ProgramFiles}/OpenBlas)
endif()

if(EXISTS ${BLAS_PACKAGE_DIR}/lib/libopenblas.lib)
    message(STATUS "Found OpenBlas in ${BLAS_PACKAGE_DIR}")
    set(BLAS_DLLS libopenblas.dll)
    set(BLAS_DLL_DIR ${BLAS_PACKAGE_DIR}/bin)
    set(BLAS_INCLUDE_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/include/)
    set(BLAS_LIB_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/lib/)
    set(BLAS_FOUND TRUE)
endif()
```
and re-run this command line in that folder:
```
cmake -G "Visual Studio 15 2017 Win64"
```
Hopefully you should see this output:
```
-- ProgramW6432=[C:\Program Files]
-- Found OpenBlas in C:\Program Files/OpenBlas
```

Ry7en, 2019-03-13:
The first line is present:

`-- ProgramW6432=[C:\Program Files]`

 but not the second:

```
C:\Users\HP\Documents\foo>cmake -G "Visual Studio 15 2017 Win64"
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- ProgramW6432=[C:\Program Files]
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/HP/Documents/foo
```

lovettchris, 2019-03-13:
Ok, great, keep adding "message" statements until we figure out what is not working here, is it the first if block that is failing?  If so this would indicate that "C:\Program Files\OpenBlas" does not exist.  (Ignore the forward slashes versus backslashes - cmake uses forward slashes but can also handle the backslashes that Windows uses).
```
if(${CMAKE_SYSTEM_PROCESSOR} STREQUAL "AMD64")
    set(BLAS_PACKAGE_DIR $ENV{ProgramW6432}/OpenBlas)
else()
    set(BLAS_PACKAGE_DIR $ENV{ProgramFiles}/OpenBlas)
endif()
```
or is it the second if block?  
```
if(EXISTS ${BLAS_PACKAGE_DIR}/lib/libopenblas.lib)
    message(STATUS "Found OpenBlas in ${BLAS_PACKAGE_DIR}")
    set(BLAS_DLLS libopenblas.dll)
    set(BLAS_DLL_DIR ${BLAS_PACKAGE_DIR}/bin)
    set(BLAS_INCLUDE_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/include/)
    set(BLAS_LIB_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/lib/)
    set(BLAS_FOUND TRUE)
endif()
```
For example, let's add this status message just before the above if-test:

```
message(STATUS "BLAS_PACKAGE_DIR=[${BLAS_PACKAGE_DIR}]")
```

Ry7en, 2019-03-13:
When CMakeLists.txt is modified to contain:

```
# This is the default location for a manually installed OpenBlas build.
message(STATUS "ProgramW6432=[$ENV{ProgramW6432}]")
if(${CMAKE_SYSTEM_PROCESSOR} STREQUAL "AMD64")
    set(BLAS_PACKAGE_DIR $ENV{ProgramW6432}/OpenBlas)
else()
    set(BLAS_PACKAGE_DIR $ENV{ProgramFiles}/OpenBlas)
endif()
message(STATUS "BLAS_PACKAGE_DIR=[${BLAS_PACKAGE_DIR}]")
if(EXISTS ${BLAS_PACKAGE_DIR}/lib/libopenblas.lib)
    message(STATUS "Found OpenBlas in ${BLAS_PACKAGE_DIR}")
    set(BLAS_DLLS libopenblas.dll)
    set(BLAS_DLL_DIR ${BLAS_PACKAGE_DIR}/bin)
    set(BLAS_INCLUDE_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/include/)
    set(BLAS_LIB_SEARCH_PATHS ${BLAS_PACKAGE_DIR}/lib/)
    set(BLAS_FOUND TRUE)
endif()
```

the output is:
```
C:\Users\HP\Documents\foo>cmake -G "Visual Studio 15 2017 Win64"
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- ProgramW6432=[C:\Program Files]
-- BLAS_PACKAGE_DIR=[C:\Program Files/OpenBlas]
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/HP/Documents/foo
```

lovettchris, 2019-03-13:
Ok, great, then we have narrowed it down to this line is failing:
```
if(EXISTS ${BLAS_PACKAGE_DIR}/lib/libopenblas.lib)
```
So can you check the contents of your folder: `C:\Program Files\OpenBlas\lib` and see if it contains a file named "libopenblas.lib" ?

Ry7en, 2019-03-13:
The only file in that location that I can see is "openblas.lib", not "libopenblas.lib".

lovettchris, 2019-03-13:
Interesting.  So OpenBlas build does something different on your machine then.  Can you run this from a command prompt window and send post the output?
```
dir /s /b "c:\Program Files"\openblas
```

lovettchris, 2019-03-13:
Hey do you have Visual Studio installed and did you built OpenBlas from a Visual Studio Developer Command Prompt ?  I see this in the OpenBlas CMakeLists.txt and it seems that this naming difference happens if cmake is not finding Microsoft Visual C++:
```
if(MSVC)
set(OpenBLAS_LIBNAME libopenblas)
else()
set(OpenBLAS_LIBNAME openblas)
endif()
```

Ry7en, 2019-03-13:
I have Visual Studio installed as per the tutorial (I recall that I unselected something like "Tests for Google" to save HDD space as I didn't think it looked like a crucial component), though I wasn't certain what was meant under "CMake 3.8"; I ended up adding the location mentioned: 

`c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe`

to my Path, in environment variables, which seemed to allow cmake to be run from the anaconda prompt.

lovettchris, 2019-03-13:
this is what I see:
```
(base) C:\Users\clovett>dir /s /b "c:\Program Files"\openblas
c:\Program Files\openblas\bin
c:\Program Files\openblas\include
c:\Program Files\openblas\lib
c:\Program Files\openblas\bin\libopenblas.dll
c:\Program Files\openblas\include\cblas.h
c:\Program Files\openblas\include\f77blas.h
c:\Program Files\openblas\include\openblas_config.h
c:\Program Files\openblas\lib\libopenblas.lib
```
I normally setup my command prompt environment using this trick:
```
call "c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\Tools\VsDevCmd.bat" 
```

Ry7en, 2019-03-13:
I see:

```

C:\Users\HP>dir /s /b "c:\Program Files"\openblas
c:\Program Files\openblas\include
c:\Program Files\openblas\lib
c:\Program Files\openblas\share
c:\Program Files\openblas\include\openblas
c:\Program Files\openblas\include\openblas\cblas.h
c:\Program Files\openblas\include\openblas\f77blas.h
c:\Program Files\openblas\include\openblas\openblas_config.h
c:\Program Files\openblas\lib\openblas.lib
c:\Program Files\openblas\share\cmake
c:\Program Files\openblas\share\cmake\OpenBLAS
c:\Program Files\openblas\share\cmake\OpenBLAS\OpenBLASConfig.cmake
c:\Program Files\openblas\share\cmake\OpenBLAS\OpenBLASConfigVersion.cmake
c:\Program Files\openblas\share\cmake\OpenBLAS\OpenBLASTargets-release.cmake
c:\Program Files\openblas\share\cmake\OpenBLAS\OpenBLASTargets.cmake
```

Thank you for the tip; should I remove the location I added to my Path and run the call?

lovettchris, 2019-03-14:
yep, let's remove this openblas install from "c:\Program Files\openblas", and your openblas/build folder, open a Visual Studio Developer Command Prompt, should look like this:
[image]
and rebuild open blas from there.  This should build openblas with MSVC and you should see the names libopenblas.  Then the ELL build should work witht that.

Ry7en, 2019-03-14:
May I simply delete the "c:\Program Files\openblas" install, or is there an uninstall procedure to be followed?

lovettchris, 2019-03-14:
you can simply delete it :-)

Ry7en, 2019-03-14:
Excellent - now there's no shame in confessing to have been a little faster in deleting than in asking questions; alas my fortune didn't endure, as after deleting the folders and following these steps:

clone https://github.com/xianyi/OpenBLAS, then:
```
mkdir build
cd build
cmake -G "Visual Studio 15 2017 Win64" ..
cmake --build . --config Release
```

I can only see a "openblas.lib" in "C:\Users\HP\OpenBLAS\build\lib\RELEASE". I did notice after searching my computer that there are "libopenblas.dll.a" files in "C:\Users\HP\ELL\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\lib" and the "x64\sandybridge\lib" folders, if that is of any importance.

lovettchris, 2019-03-14:
Hmmm, very odd, now I need to compare cmake output from your OpenBLAS folder with what I'm getting.  Can you run this?
```
cmake -G "Visual Studio 15 2017 Win64" .. > cmake.log 2>&1
```
You should get something like this:
[cmake.log](https://github.com/Microsoft/ELL/files/2968466/cmake.log)
And the build should finish with output like this:
```
  libopenblas.vcxproj -> D:\git\OpenBLAS\build\lib\RELEASE\libopenblas.dll
```

Ry7en, 2019-03-14:
When I ran `cmake -G "Visual Studio 15 2017 Win64" .. > cmake.log 2>&1
` from my  "C:\Users\HP\OpenBLAS\build" folder, "cmake.log" was generated, but the command prompt didn't output details of a build. 

[cmake.zip](https://github.com/Microsoft/ELL/files/2968788/cmake.zip)

lovettchris, 2019-03-14:
Thanks, very different output from mine.  I think you may need to ask for some help on the OpenBlas git repo to figure out how to get OpenBlas to build properly using the MSVC compiler.  See how my cmake output in yellow right off the bat identifies CXX is MSVC version 19.16... yours doesn't.
[image]

lovettchris, 2019-03-14:
You could try doing this before running the OpenBLAS cmake:
```
call "c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\Tools\VsDevCmd.bat" 
set WindowsSdkDir=
```
And delete the OpenBLAS build folder each time you rerun cmake to make sure it is a clean cmake generation from scratch.

Ry7en, 2019-03-15:
I ran `call c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\Tools\VsDevCmd.bat"` before generating the cmake.log attached in my previous post, but not `set WindowsSdkDir=`. Running both and rebuilding I noticed this line at the beginning of the new log "The C compiler identification is MSVC 19.16.27027.1" followed by others similar to the ones in your cmake.log.

[cmake 2.zip](https://github.com/Microsoft/ELL/files/2968911/cmake.2.zip)

lovettchris, 2019-03-15:
Looks much better.  This is what I mean by building OpenBLAS from a "Visual Studio Developer Command Prompt".  You should be good to go now.

Ry7en, 2019-03-17:
Unfortunately the next step

` cmake --build . --config Release`

fails to produce a "libopenblas.dll", only openblas.lib as before. I've opened an issue on the OpenBLAS repository as suggested, so with any luck will soon get to the bottom of this.

Ry7en, 2019-03-18:
The solution, as offered to me in the OpenBLAS issue forum, seems to be running:

`cmake -DBUILD_SHARED_LIBS=ON ..`

before:

`cmake --build . --config Release`

The "RELEASE" folder now contains openblas.dll, as well as openblas.lib, openblas.exp and openblas.pdb, and one can apparently rename openblas to libopenblas if that is what ELL expects. 
In order to add python bindings, I ran `cmake --build . --target _ELL_python --config Release` from the OpenBLAS\build folder, giving this output:

```
MSBUILD : error MSB1009: Project file does not exist.
Switch: _ELL_python.vcxproj
```
Given that I ran `cmake -G "Visual Studio 15 2017 Win64" ..` from the VS developer prompt, am I right to assume this is due to not following this direction in the installation tutorial:

"If you intend to build Python interfaces, make sure to activate the py36 environment as described above."?

Ry7en, 2019-03-26:
The final hurdle proved to be the location of `cblas.h`, `f77blas.h` and `openblas_config.h`; instead of being located in "C:\Program Files\openblas\include\" as OpenBLASSetup.cmake expected, they resided in "C:\Program Files\openblas\include\openblas\". Once I had copied the three files into the correct folder, ELL built without a hitch.

lovettchris, 2019-03-26:
hey congrats, that's great.