microsoft/ELL #202: Error during build ELL on ubuntu:  error: ‘outputLayout’ is not captured

jaehyunshinMLS, 2019-03-12:
Hi,

I am trying to set up ELL on ubuntu 18 by following the instruction.

After install requirements, try to build with cmake ..

The error came up :  error: ‘outputLayout’ is not captured 

ELL/libraries/nodes/include/BroadcastOperationNodes.h:515:58: error: ‘outputLayout’ is not captured
         const auto outputGlobalOffset = static_cast<int>(outputLayout.GetFirstEntryOffset())
Do you have any idea?

lovettchris, 2019-03-12:
weird, static_cast is not a lambda, so there's nothing to capture.  But the previous line tries to do this "const auto&" to a return type of type "model::PortMemoryLayout" which also doesn't really make sense, so perhaps try replacing line 514 with this:
```cpp
model::PortMemoryLayout outputLayout = GetOutputMemoryLayout();
```

lovettchris, 2020-12-02:
The latest bits have been tested on Ubuntu 18 and are building cleanly with gcc-8 and clang-8.

closed 2020-12-02 by lovettchris